xoseperez/espurna #465: Auto discover feature with Home Assistant ?

vincegre, 2018-01-25:
Hi

It's listed both in repository and wiki that it supports auto discover MQTT feature of Home Assistant but I couldn't find any explanations anywhere (or was not able to find it) about how to configure that on ESP side ! I have already done config in Home Assistant !

Thanks for help ;)

Vincèn

xoseperez, 2018-01-25:
Hi @vincegre

The MQTT auto discover feature works for switches and lights. You have to have Home Assistant connected to the same MQTT broker as the ESP and then just enable the Home Assistant option in the General tab of the web UI. If you have not configured otherwise the default prefix "homeassistant" should work.

Once you enable it (and click on the "Save" button) it will send an MQTT message to a topic like "homeassistant/switch/ESPURNA_XXXXXX/config" with the required information so HA can configure the switch. Same for lights. 

Yes, there is a lot of work to be done on the docs, yet...

vincegre, 2018-01-26:
Hi @xoseperez 

Thanks a lot for details and great to know it works in so easy way so going to test it today ;) No support for sensors in automatic too ?

For docs I would be happy to help if I can in any way ;)

Vincèn

xoseperez, 2018-01-26:
I'm working on adding auto-discover for sensors too. 

Any help is very welcome. If you want to contribute to the documentation clone the wiki (see the link below the table of contents), modify it and submit a pull request.

vincegre, 2018-01-26:
> I'm working on adding auto-discover for sensors too.

Cool ;)

> Any help is very welcome. If you want to contribute to the documentation clone the wiki (see the link below the table of contents), modify it and submit a pull request.

Will check how to do that and send you hopefully soon some pull requests for documentation ;)

xoseperez, 2018-01-26:
@vincegre I have just pushed a commit to dev branch with support for auto-discover sensors and multi-relays boards (forgot to mention the issue on the commit, but here it is: cd54372588342d949e13f1800a7bad9532a38cf9).

Maybe you could give it a try and report back.

vincegre, 2018-01-26:
@xoseperez Thanks a lot for it and definitively going to report how it's going as soon as it's flashed ;) Do you have a way to generate the binary file to download ? as I gave up long time ago to succeed to compile locally ESP firmwares :(

xoseperez, 2018-01-26:
That's bad :)

I can share a wetransfer with the image. Any specific device?

vincegre, 2018-01-26:
I know it's bad and I should take some time to setup a proper environment to be able to build them easily ;)

Will use relay, DHT shield, and BMP180 sensors ;) Thanks a lot !

vincegre, 2018-01-26:
Succeeded to compile here with Atom and Platform.io setup in it :) Just completely lost how I can compile with the commit you made for autodiscover for sensors ? :/

xoseperez, 2018-01-26:
That's good 😄 

You have to checkout the "dev" branch. It will depend (slightly) on the OS you use but to change branch from the terminal you will have to type:

`git checkout dev`

If you are using a git frontend I can't tell you the steps but the idea is the same: to checkout the dev branch.

Once you are in the dev branch you build it and flash it just like before.

closed 2018-01-28 by xoseperez

brandon250r, 2018-02-15:
It seems my homeassistant button on the general page will not stay enabled after I click save. When I refresh the page or restart the device the button is back at off and HA is not finding anything. I did notice in my mqtt logs that the device is constantly connecting and disconnecting. I may just reflash and try again, unless someone has a suggestion?

vincegre, 2018-02-15:
`It seems my homeassistant button on the general page will not stay enabled after I click save. When I refresh the page or restart the device the button is back at off and HA is not finding anything. I did notice in my mqtt logs that the device is constantly connecting and disconnecting. I may just reflash and try again, unless someone has a suggestion?`

For HA button it's normal behaviour as once automatic discovery has been done it doesn't need it at each restart !
For your MQTT logs do you mind to share it here to determine what's the problem ?
Did you check also that you erased Flash before flashing ? and that your power supply is enough powerful ? as it's often source of most problems !

vincegre, 2018-07-29:
@jsponz If I remember it was working but I don't use anymore Espurna since ages !! Found an other one lot more flexible, easier to use (not focused on sonoff devices): esphomeyaml :+1: