xoseperez/espurna #466: Translations procedure

xoseperez, 2018-01-25:
Define and implement a procedure to translate the web interface.

A requirement is that this feature must not add file size to the final binary image, not beyond what could be normal due to different string lengths for different languages.

Right now all "translatable" strings are either on the index.html or the custom.js files. The idea would be to have a procedure implemented in the gulpfile.js that will search and replace the strings from localization files. There are a number of gulp plugins that could provide this solution.

fork04, 2018-01-25:
Link doesn`t work
https://github.com/xoseperez/espurna/issues/455#issuecomment-360145421

xoseperez, 2018-01-26:
I just fixed the link, thanks.