xoseperez/espurna #468: custom.h include order

mcspr, 2018-01-25:
Using dev branch, *custom.h* is now defined before the default headers.
Not all definitions are guarded by ifndefs, so when custom header got settings like `ADMIN_PASS`, `NTP_SERVER`, `MQTT_TOPIC` they get overridden back to default ones.

xoseperez, 2018-01-25:
There are two different options here: parse custom.h first and use #ifndef everywhere in general.h or parse custom.h last and used #undef before to avoid warnings by the compiler.

Problem with the second option is that some definitions depend on others, even inside the same general.h file. I don't know if this is actually a design error but first option above looks better. That, of course, means adding #ifndef guards everywhere. Some of the settings already have them, basically those I had to redefine at some point.

mcspr, 2018-01-26:
My current custom.h (without couple of _SUPPORT defines):
```
#define ADMIN_PASS=...
#define WEB_FORCE_PASS_CHANGE=0
#define WIFI_SCAN_NETWORKS=0
#define NTP_SERVER=...
#define NTP_TIME_OFFSET=0
#define MQTT_TOPIC="prefix/{identifier}"
```

I would agree that first option is better. Plus with ifndefs there is possibility to also use PLATFORMIO_BUILD_FLAGS to sometimes change settings per build (debug, pins etc.)

lobradov, 2018-02-07:
Trying to figure this out is like playing "Rock paper scissors lizard Spock" :) 

Rules we have: 
- command line preprocessor flags get processed first
- defaults should set the general scene before "customizations". 
- customization should override defaults, but not command flags 
- command line flags should have the highest preference, overriding anything else
- dependency resolution should come last to make config consistent

I can think of a very complicated set of intermediary flags to signal if parameter was defined in command line or from default, and how should later steps be taken, but that would make code unreadable beyond any recognition.

Other option would be to restrict the use of command line to only point to custom configuration (`-Iconfig/my_config.h`), that would then be processed after defaults, but before dependency resolving set of macros... 

In either case, it would make sense to split types from defaults and dependency resolving macros, and have them in separate files for easier manipulation. 

@xoseperez - I can give it a try if you agree with overall approach.

xoseperez, 2018-02-08:
Yes, it makes sense. The general.h file is a mixture of everything (types, defaults, dependencies) and it sure needs some cleaning up. I started some time ago splitting the hardware.h file into definitions (current hardware.h) and defaults (defaults.h) file.

@lobradov If you want to propose a new structure for the header files go ahead.

PieBru, 2018-02-12:
Hi, I had this same issue trying to enable WEB_SUPPORT in ESPURNA_CORE.
Adding WEB_SUPPORT in custom.h does not work and obviously gives build warnings, so I think a quick temporary solution could be rename the first custom.h as custom_defaults.h and add to the bottom of all.h a custom_config.h, both under #ifdef USE_CUSTOM_H
```
#ifdef USE_CUSTOM_H
#include "custom_defaults.h"
#endif

#include "version.h"
...

#ifdef USE_CUSTOM_H
#include "custom_config.h"
#endif
```

xoseperez, 2018-04-18:
Closing this in favour of #764. Please join the conversation there.

closed 2018-04-18 by xoseperez