xoseperez/espurna #469: RF remote compatibility / Arilux versions

Fietspomp86, 2018-01-25:
I have a question, do you think it’s possible to add support for more versions of the Arilux controller?
Also I think these are the same as the Magic Home controllers?
Here’s the version specific info I found:
 
**AL-LC01 - No Controller, RGB, 5-28V
AL-LC02 - No Controller, RGBW, 9-12V
AL-LC03 - iR Controller, RGB, 5-28V
AL-LC04 - iR Controller, RGBW, 9-12V
AL-LC08 - No Controller, RGBWW, 5-28V
AL-LC09 - RF Controller, RGB, 5-28V
AL-LC10 - RF Controller, RGBW, 9-28V
AL-LC11 - RF Controller, RGBWW, 9-28V**

Flashed my AL-LC03 with custom compiled Espurna, IR is working fine.
Did not try to implement the RF yet.

Saw on Github somebody already made different firmware with RF support, maybe it’s possible to copy this code (https://github.com/mertenats/Arilux_AL-LC0X)? If permission is granted ofcourse,

johny-mnemonic, 2018-02-09:
I can help with testing if you decide to support this.
I have Magic Home variant matching AL-LC10 (RGBW with RF remote).

jonny190, 2018-02-14:
I also have a LC10 but cannot get the RF remote working

johny-mnemonic, 2018-02-15:
@jonny190 not sure what FW you are trying to use for this, but generally it is about finding on which GPIO pin the RF receiver data connection is and then also on which GPIO pin the receiver control pin is connected. You can [check ](https://github.com/arendst/Sonoff-Tasmota/issues/1338#issuecomment-364703158) what I needed to do to make it working under Tasmota.

gn0st1c, 2018-04-28:
aha! this explains why my pins does not match with the pins defined in espurna
[IR_Remotes.zip](https://github.com/xoseperez/espurna/files/1957628/IR_Remotes.zip)
.

> WARNING: DUE TO A NEW PINOUT, THIS FIRMWARE SEEMS TO BE NOT MORE COMPATIBLE WITH THE NEWEST MODELS (PCB version > 1.4).

i received 3 different IR remotes with these devices. here are the photos & IRremoteESP8266 codes.

hurzhurz, 2018-08-10:
I have made a try attempt to implement RF remote support for the LC-11.
It's incomplete any maybe a bit crude, but it does work so far.
Maybe it can be used as a starting point for a clean implementation?

[espurna-1.13.1_lc11-rf.patch.txt](https://github.com/xoseperez/espurna/files/2279144/espurna-1.13.1_lc11-rf.patch.txt)
Speed and Mode are not implemented.
Mode+/- are misused for color temperature.
And as the codes can be different between multiple remotes, I have made the variable "home code" part changeable as a setting "rfHomeCode".