xoseperez/espurna #470: quinled not working

tarbax, 2018-01-25:
i cant control the two outputs nothing changes when i dim the led or want to shutoff.
1output has power the other not. i uploaded true arduino ide and selected intermittech quinled in the config folder with support for mqtt and home assistant.

xoseperez, 2018-01-26:
Do you have access to the debug log? What version of the QuinLED are you using?

I don't own any QuinLED. Support for QuinLED in ESPurna was provided by @ColinShorts based on his findings with version 2.6. Check Intermit.tech post about it here: http://blog.quindorian.org/2017/02/esp8266-led-lighting-quinled-v2-6-pcb.html/

ColinShorts, 2018-01-26:
I've built a few of these now, and not really had many problems with the esuprna firmware for QuinLED, on the board itself, or on a separate ESP-01 mounted inside a light with a mosfet.

What version of the firmware are you using - I'm happy to test over the weekend.

Did you test that the board itself is working ok? From my experience if you power the whole thing off and pull the ESP module the mosfets should switch on both outputs fully - if that isn't the case then you might have a problem with the board itself - possibly not using low voltage mosfets, or a problem with the soldering itself. It might also be worth double-checking the voltage is set correctly (~3.3-3.5V should be fine, but pull the ESP module first so as not to release the blue smoke).

@xoseperez - I could mail one or two over to you if you want - but I'll let you supply your own ESP modules as I'm running a bit low on them atm ;)

xoseperez, 2018-01-26:
@ColinShorts thanks for your offering. I'm surrounded by stuff all around. I don't really need more devices to test :) 

I think we should wait for @tarbax  to report back. It could be a bad flash or a different device version.

ColinShorts, 2018-01-26:
@xoseperez - I feel/share your pain 😉 

@tarbax - waiting on you 😃

onlytobeking, 2018-01-26:
I'm having the same problem as @tarbax, tried with two different quinleds and mosfets from two different suppliers but still no response, last try with the latest version of ESPurna. @ColinShorts so if I power it off, remove the esp, power it back on it should be fully on on both channels?

tarbax, 2018-01-26:
@onlytobeking @ColinShorts i have one working i tink i have a bad mosfet on the first board. if the esp is out the socket and you power it on it should power both channels.

onlytobeking, 2018-01-26:
Is it possible I have four bad mosfets from two different suppliers?
The only thing else is I skipped the tiny tiny resistors on the board and soldered the pads together as in previous versions of the quinleds. That shouldn't be a problem, should it?

ColinShorts, 2018-01-26:
@onlytobeking - Those resistors are present on all of my assembled boards - kinda fiddly, but tinning one of the contacts on the board, placing the resistor on top and touching it with the iron should be enough to hold it in place to solder the other end (three hands optional, tweezers or a toothpick recommended).

I don't think that the smd resistors are needed for normal operation - only for flashing the esp module, but I could be wrong. With the mosfets, it's possible/likely that you've got units that switch at 5V rather than the ones recommended which should switch happily at 3.3V. Can you check the voltage at the mosfets with the dimmer set to zero and then max?

@tarbax - yes, both should come on with the esp module removed, so it does sound like a faulty mosfet. Is it v2.6 of the board - and what version of the firmware are you using (or are you building from source?)

I've been using platformio to build/flash a variety of units (aithinker, quinled, sonoff T1's, dual Rev2, H801's, etc), and have found that flashing again fixes any problems that aren't hardware related, or were caused by a bad ota flash or other randomness.

onlytobeking, 2018-01-26:
@ColinShorts yep, I'm up for the challenge to solder them just forgot to order them when I was ordering all of the other stuff and I was eager to get started, therefore I skipped them. I'm using one of those usb-serial-dock thingies to flash them so I'm not really flashing the esp when its attached to the quinled. I'm gonna give it another try this weekend, try flashing it again and try to measure the mosfets. I´ll get back to you. Thanks for your help.

ColinShorts, 2018-01-26:
@onlytobeking - it might be worth trying an older binary in case the latest has introduced some weirdness - but I'll give the latest a shot myself over the weekend and report back.

tarbax, 2018-01-26:
@onlytobeking maybe you are heating the mosfets up to breaking point when soldering. cause that would be very unlikely that you have so many bad one's. i like to hold the mosfet up with my finger and when i feel its getting hot ill stop for a second and blow on it. just got the third one working to. now figuring home assistant config out. seems that i am dumb today cant get it working for some reason. hard configs are easy for me and easy configs are to hard :p got espurna on my lights.

onlytobeking, 2018-01-26:
@tarbax nice work! Don’t they automatically show up in homeassistant? Think one of mine showed up even though it wasn’t doing anything. The thought about the heat actually crossed my mind when I was soldering, if nothing else works I’ll solder another one and be super careful, thanks for the tip. Hope you’re up and running with everything in home assistant soon.

ColinShorts, 2018-01-27:
For what it's worth, I've just flashed a precompiled 1.12.1 using OTA and it seems to be working fine.

onlytobeking, 2018-01-29:
Tried to measure the mosfets this weekend and it seems like they have 3,3 v no matter what setting the sliders in ESPurna has.
I'm flashing the firmware with platform IO. Just to make sure, is there anything else I need to change other than the env_default?

xoseperez, 2018-01-29:
No need for that either, you can compile and flash (wired) any environment like this:

`platformio run -e intermittech-quinled -t upload`

ColinShorts, 2018-01-29:
For platformio I think you'll just need to use:
$ pio run -e intermittech-quinled -t upload

-Colin

onlytobeking, 2018-01-30:
Finally success, I soldered a new board and got one channel working. So probably never was a problem with the ESPurna, just my shitty soldering. Thank you all for your patience and help!

xoseperez, 2018-01-30:
👍 No problem, it's good you solved it. Closing

closed 2018-01-30 by xoseperez

tarbax, 2018-01-30:
can anyone give there home assistant config for manual mqtt cause min will not work:

my config:
light:
  - platform: mqtt
    name: 'quinled keuken'
    state_topic: '/quinled01/relay/0'
    command_topic: '/quinled01/relay/0/set'
    payload_on: 1
    payload_off: 0
    brightness: true
    brightness_scale: 255
    brightness_command_topic:  '/quinled01/brightness/set'
    brightness_state_topic: '/quinled01/brightness'

rqleslie, 2019-10-02:
@tarbax did you ever come right the YAML to get this working, Im trying to get a slider to work in Home Assistant to control the dimming / on/off with a QuinLED-OG