xoseperez/espurna #471: pre-compiled bin files?

myounges, 2018-01-25:
Hello,

First of all thank you very much for your work, I really appreciate your efforts and fully support this initative to switch from bitbucket to github.

Not sure if this is the right channel to ask, but I was wondering if this repository will have a pre-compiled binary library as the old repository had, I found it very useful for the upgrade feature.

Keep up the great work.
Adeu,

nkerezov, 2018-01-25:
They are there  at the top bottons

myounges, 2018-01-25:
Hello sorry when I to said binaries I meant the firmware bin files for each hardware model what was in the downloads section.

Are we talking of the same files? Wasn't able to find them.

nkerezov, 2018-01-25:
Yes,
Ill try to make it easyer since Im using the same method as you and it was the first think I was looking for ;-) 
https://github.com/xoseperez/espurna/releases

myounges, 2018-01-25:
So embarrassing... Didn't see them. Thank you nkerezov!!!

nkerezov, 2018-01-25:
Dont worry, I had the same panic atack......

xoseperez, 2018-01-26:
Thanks @nkerezov !

closed 2018-01-26 by xoseperez