xoseperez/espurna #472: Sonoff TH temp+hum in Domoticz

nkerezov, 2018-01-25:
Hi guys,
Can anyone share the secret ,how to make hum+temp from Sonoff TH on one "ticket" in Domoticz?
I can make one for hum and one for temp ,but no idea how to build them at one place ,so I can get my space organized and have the dew point calculated.
Cheers!

xoseperez, 2018-01-26:
Is that something that should be done from the ESPurna point of view or inside the Domoticz configuration?

nkerezov, 2018-01-26:
I think if you can combine temp and hum i one under one IDX it will somehow work.In espeasy works like this ,but I still like espurna way more ,because it is super fast on boot and even if its disconected from the brocker it works like a charm ( not the case with espeasy)
Cheers!