xoseperez/espurna #473: Save trouble on 1.12.2b

chryss44, 2018-01-25:
Hi,
Just for information
I have tried to compile 1.12.2b (sonoff basic)and have many trouble to save parameters, like new AP Wifi or other and when i restart, all parameters return default.

chryss44, 2018-01-25:
I have just compiled 1.12.2b and put it on Sonoff Basic.
After First restart, i have put new password, add Wifi AP and save.
I have just click on reboot and all parameter return to default.
The log is : `[000084] 

[000084] [INIT] ESPURNA 1.12.2b
[000084] [INIT] xose.perez@gmail.com
[000084] [INIT] http://tinkerman.cat

[000084] [INIT] CPU chip ID: 0x8A6B66
[000087] [INIT] CPU frequency: 80 MHz
[000090] [INIT] SDK version: 1.5.3(aec24ac9)
[000094] [INIT] Core version: 2_3_0_RC2
[000097] [INIT] Core revision: 
[000100] 
[000101] [INIT] Flash chip ID: 0x14405E
[000104] [INIT] Flash speed: 40000000 Hz
[000108] [INIT] Flash mode: DOUT
[000111] 
[000112] [INIT] Flash sector size:     4096 bytes
[000116] [INIT] Flash size (CHIP):  1048576 bytes
[000120] [INIT] Flash size (SDK):   1048576 bytes /  256 sectors
[000126] [INIT] Firmware size:       473740 bytes /  116 sectors
[000132] [INIT] OTA size:            573440 bytes /  140 sectors
[000137] [INIT] SPIFFS size:              0 bytes /    0 sectors
[000143] [INIT] EEPROM size:           4096 bytes /    1 sectors
[000148] [INIT] Empty space:          16384 bytes /    4 sectors
[000154] 
[000155] [INIT] BOARD: ITEAD_SONOFF_BASIC
[000159] [INIT] SUPPORT: ALEXA BROKER DEBUG_SERIAL DEBUG_TELNET DOMOTICZ HOMEASSISTANT MDNS_SERVER NTP SCHEDULER SSDP TELNET TERMINAL THINGSPEAK WEB

[000172] [INIT] Last reset reason: Power on
[000175] [INIT] Settings size: 58 bytes
[000179] [INIT] Free heap: 28320 bytes
[000183] [INIT] Power: 4094 mV
[000185] [INIT] Power saving delay value: 10 ms
[000189] 
[000196] [TELNET] Listening on port 23
[000197] [WEBSERVER] Webserver running on port 80
[000200] [RELAY] Retrieving mask: 215
[000201] [RELAY] Relay #0 boot mode 0
[000204] [RELAY] #0 set to OFF
[000208] [RELAY] Number of relays: 1
[000211] [BUTTON] Number of buttons: 2
[000214] [LED] Number of leds: 1
[000217] [MQTT] Async ENABLED, SSL DISABLED, Autoconnect ENABLED
[000228] [SSDP] Started
[000230] [THINGSPEAK] Async ENABLED, SSL DISABLED
[000238] [MAIN] Uptime: 0 seconds
[000238] [MAIN] Free heap: 22376 bytes
[000238] [MAIN] Power: 4094 mV
[000259] [WIFI] Creating access point
[000260] [WIFI] MODE AP --------------------------------------
[000260] [WIFI] SSID  ESPURNA_8A6B66
[000261] [WIFI] PASS  fibonacci
[000263] [WIFI] IP    192.168.4.1
[000266] [WIFI] MAC   6A:C6:3A:8A:6B:66
[000270] [WIFI] ----------------------------------------------
[000276] [MDNS] OK
[001207] [RELAY] Saving mask: 0
[007160] [WEBSERVER] Request: GET /index.html
[007758] [WEBSOCKET] #1 connected, ip: 192.168.4.2, url: /ws
[026396] [WEBSOCKET] Parsing configuration data
[026417] [NTP] Time: 
[029492] [WEBSERVER] Request: GET /index.html
[035399] [WEBSERVER] Request: GET /index.html
[035634] [WEBSOCKET] #1 disconnected
[036169] [WEBSOCKET] #2 connected, ip: 192.168.4.2, url: /ws
[060008] [MAIN] System OK
[070475] [WEBSOCKET] Parsing configuration data
[081349] [WEBSOCKET] Requested action: reboot

 ets Jan  8 2013,rst cause:1, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
ve64e32b3
~ld
⸮[000084] 

[000084] [INIT] ESPURNA 1.12.2b
[000085] [INIT] xose.perez@gmail.com
[000085] [INIT] http://tinkerman.cat

[000085] [INIT] CPU chip ID: 0x8A6B66
[000087] [INIT] CPU frequency: 80 MHz
[000090] [INIT] SDK version: 1.5.3(aec24ac9)
[000094] [INIT] Core version: 2_3_0_RC2
[000098] [INIT] Core revision: 
[000100] 
[000101] [INIT] Flash chip ID: 0x14405E
[000105] [INIT] Flash speed: 40000000 Hz
[000108] [INIT] Flash mode: DOUT
[000111] 
[000112] [INIT] Flash sector size:     4096 bytes
[000117] [INIT] Flash size (CHIP):  1048576 bytes
[000121] [INIT] Flash size (SDK):   1048576 bytes /  256 sectors
[000126] [INIT] Firmware size:       473740 bytes /  116 sectors
[000132] [INIT] OTA size:            573440 bytes /  140 sectors
[000138] [INIT] SPIFFS size:              0 bytes /    0 sectors
[000143] [INIT] EEPROM size:           4096 bytes /    1 sectors
[000149] [INIT] Empty space:          16384 bytes /    4 sectors
[000154] 
[000156] [INIT] BOARD: ITEAD_SONOFF_BASIC
[000159] [INIT] SUPPORT: ALEXA BROKER DEBUG_SERIAL DEBUG_TELNET DOMOTICZ HOMEASSISTANT MDNS_SERVER NTP SCHEDULER SSDP TELNET TERMINAL THINGSPEAK WEB

[000172] [INIT] Last reset reason: Software/System restart
[000177] [INIT] Settings size: 58 bytes
[000181] [INIT] Free heap: 28320 bytes
[000185] [INIT] Power: 4094 mV
[000187] [INIT] Power saving delay value: 10 ms
[000191] 
[000198] [TELNET] Listening on port 23
[000199] [WEBSERVER] Webserver running on port 80
[000202] [RELAY] Retrieving mask: 215
[000203] [RELAY] Relay #0 boot mode 0
[000206] [RELAY] #0 set to OFF
[000210] [RELAY] Number of relays: 1
[000212] [BUTTON] Number of buttons: 2
[000216] [LED] Number of leds: 1
[000218] [MQTT] Async ENABLED, SSL DISABLED, Autoconnect ENABLED
[000230] [SSDP] Started
[000232] [THINGSPEAK] Async ENABLED, SSL DISABLED
[000239] [MAIN] Uptime: 0 seconds
[000239] [MAIN] Free heap: 22376 bytes
[000240] [MAIN] Power: 4094 mV
[000261] [WIFI] Creating access point
[000262] [WIFI] MODE AP --------------------------------------
[000262] [WIFI] SSID  ESPURNA_8A6B66
[000262] [WIFI] PASS  fibonacci
[000265] [WIFI] IP    192.168.4.1
[000268] [WIFI] MAC   6A:C6:3A:8A:6B:66
[000271] [WIFI] ----------------------------------------------
[000277] [MDNS] OK
[001209] [RELAY] Saving mask: 0
[004819] [WEBSERVER] Request: GET /index.html
[029747] [WEBSERVER] Request: GET /index.html
`

xoseperez, 2018-01-26:
Can you do a couple of tests?

First thing you can do is to type "keys" on the terminal screen just after changing and saving the settings in the web interface and before rebooting the device. It should show you the current settings (what you just have saved). If after reboot you type "keys" again and it your settings have disappeared there is indeed a problem.

The second test is trying an old version with the same device. This will determine if there is a problem with the current version.

You could also test a simple EEPROM script to discard an issue with the hardware:

```
#include <EEPROM.h>

void setup() {
    Serial.begin(115200);
    EEPROM.begin(4096);
}

void loop() {
    byte value = EEPROM.read(0);
    Serial.printf("Current value is: %d\n", value);
    EEPROM.write(0, value + 1);
    EEPROM.commit();
    delay(2000);
    ESP.restart();
}
```
This sketch should reboot every 2 seconds and print an incrementing value retrieved from the EEPROM.

chryss44, 2018-01-26:
i have try 1.12.1 and it’s OK, all parameter are save.
After reboot, all are OK.

Sorry, Xose, perhaps it is activation off ssd, no ?
 Or my compilation are not correct ?

xoseperez, 2018-01-26:
Have you changed any setting from the default ones when building you image?

chryss44, 2018-01-26:
I have just active SSDP support, would you like i try without ?

xoseperez, 2018-01-26:
That's hardly the reason... I cannot reproduce it with my testing boards, will have to search for a Basic I have somewhere.

chryss44, 2018-01-26:
I’ll’ try with other board (basic)  to test if it’s the same and with SSDP disable

xoseperez, 2018-01-28:
Moving forward so it doesn't block the release.

chryss44, 2018-02-02:
Hi Xose, 
I have tried and maked other tests and thin the problem is on Compile or tools used to program Sonoff.
I don't think is Espurna.
Yesterday i compile Sonoff Basic on 1.12.3 and all work fine with a Windows Programmer and if i use Esptools on Osx, have "save trouble" :-(

xoseperez, 2018-02-03:
I have gone through your log again and I noticed you are using Arduino Core 2.3.0_RC2. That's a pretty old version. Try upgrading, that may be the problem. Maybe you have different core versions on those two machines.

chryss44, 2018-02-09:
I have tried with 2.4 and have the same problem. I'm sure is esptools on mac because with Windows Programmer, no problem.
Juste another think, 1.12.2 are OK on SAFARI but not 1.12.3, do you want i open an issue ?

xoseperez, 2018-02-09:
There is already issue #507  about Safari problem. Maybe you can add your feedback there.

pezinek, 2018-03-17:
I hit this save issue using esptool.py and stock image espurna-1.12.4-wemos-d1mini-relayshield.bin,
when compiled myself (from dev branch) and uploaded using platformio the save button was working fine.

xoseperez, 2018-03-19:
What command are you using to flash the binary to the device? What flash mode?

pezinek, 2018-03-19:
I just copy-pasted the command from the  [wiki](https://github.com/xoseperez/espurna/wiki/Binaries) : 
```esptool.py --port /dev/ttyUSB0 --baud 115200 write_flash -fs 1MB -fm dout 0x00000 ~/Downloads/espurna-1.12.4-wemos-d1mini-relayshield.bin```

xoseperez, 2018-03-19:
The Wemos has a 4MB flash memory (`-fs 4MB`).

xoseperez, 2018-07-02:
Any update on this? Will close otherwise.

closed 2019-12-31 by mcspr