xoseperez/espurna #474: Which firmware for D1 Mini ??

vincegre, 2018-01-26:
Hi,

I'd like to flash a D1 mini board with Espurna but 3 firmwares are avalaible for d1 mini. Which one should i use ?

thanks

Vincèn

xoseperez, 2018-01-26:
The basic one the "espurna-1.12.1-wemos-d1mini-relayshield.bin", the other two have specific support for power monitor chips you can find in some devices. Check this post if you are interested: http://tinkerman.cat/connected-power-meter/

vincegre, 2018-01-26:
> The basic one the "espurna-1.12.1-wemos-d1mini-relayshield.bin", the other two have specific support for power monitor chips you can find in some devices.

Thanks for confirmation ;) Flashed with success but not usable in my case as relay is not on the I/O expected and can't change it in hardware (Little piggy modules that pile one above the other....

> Check this post if you are interested: http://tinkerman.cat/connected-power-meter/

Thanks for link but had seen it already ;)

xoseperez, 2018-01-26:
You can try to build an image with the GPIO for your hardware.

chryss44, 2018-01-26:
hi, i have choose  espurna-1.12.1-itead-sonoff-t1-2ch.bin for D1 mini pro and it work like a charm :-)

vincegre, 2018-01-26:
@chryss44 D1 Mini here :/

chryss44, 2018-01-26:
Yes, i know, and have a Wemos D1 mini pro with espurna-1.12.1-itead-sonoff-t1-2ch on it  and it work

chryss44, 2018-01-26:
[image]
This one

vincegre, 2018-01-26:
Yep but how did you setup devices connected to it ? as it has to be configured in firmware when you build it :( It can't be configured in live once loaded in D1 !

chryss44, 2018-01-26:
I prefer setup device D5 for relay 1 and D3 for button 2.

chryss44, 2018-01-26:
you can use espurna-1.12.1-itead-sonoff-t1-1ch if you need just one relay.

vincegre, 2018-01-29:
> you can use espurna-1.12.1-itead-sonoff-t1-1ch if you need just one relay.

Not working for me as I expect it :( My relay is only controled if I push second relay in interface and I discovered a bug of the Autodiscover (@xoseperez ) I guess ! It immediately reported that new device in HA as soon as I configured my Espurna D1 but it reports only one relay which is unfortunately for me the first relay in HA (second relay doesn't show unless it should !!

chryss44, 2018-01-29:
Can you decribe your hardware (relay and button) ?
It is true that I loaded the firmware espurna-1.12.1-itead-sonoff-t1-2ch
and that the first one works, I wanted to use the second later

vincegre, 2018-01-29:
> Can you decribe your hardware (relay and button) ? It is true that I loaded the firmware espurna-1.12.1-itead-sonoff-t1-2ch and that the first one works, I wanted to use the second later

In fact I use these boards https://fr.aliexpress.com/item/15PCS-WAVGAT-D1-mini-Pro-WiFi-development-board-KIT-NodeMcu-Lua-based-on-ESP8266-D1-mini/32816174727.html?spm=a2g0s.9042311.0.0.q60CiM but no worries, I'm going to compile home made version of Espurna for my hardware so I'm sure it works at best for me 👍