xoseperez/espurna #476: Espurna board

namezero, 2018-01-27:
How to reset the "Energy" value? is there an API ? thanks

xoseperez, 2018-01-27:
No, there is none. Reset the board 😄

namezero, 2018-01-27:
the board is in the wall :)
it would be nice to have this as an api service

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Garanti
sans virus. www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

namezero, 2018-01-27:
Also the Energy is growing even when relay is OFF, is that the board consumption?

xoseperez, 2018-01-27:
That and maybe "noise" captured by the HLW8012. The board can use maybe 1W (normally less than that but let's say it's 1W). That's 1 joule every second, 60J per minute. Does that match your observations?

namezero, 2018-01-27:
That's exactly my observations ! 1J/sec

xoseperez, 2018-08-27:
You can reset the energy value from the web UI since 1.12.5, see #671 .

closed 2018-08-27 by xoseperez

namezero, 2018-11-04:
Is it available from API ?