xoseperez/espurna #477: Long commands in telnet

mcspr, 2018-01-27:
I was having problem setting long WiFi and MQTT passwords using telnet module. Serial connection works fine typing same things.
Turns out, it is somehow dependant on length and stops working around 31. I inserted small debug message inside telnet/onData to display value of data and len:
```
client->onData([i](void *s, AsyncClient *c, void *data, size_t len) {
    DEBUG_MSG_P(PSTR("[TELNET] Data len: %u\n"), len);
    DEBUG_MSG_P(PSTR("[TELNET] Data msg: %s\n"), (char *) data);
    _telnetData(i, data, len);
}, 0);
```
[image]

I am a bit fuzzy where exactly this breaks.

xoseperez, 2018-01-27:
The limit is 32 and is defined by the STREAM_INJECTOR_BUFFER_SIZE constant in 'code/espurna/libs/StreamInjector.h'. Basically, it defines a ring buffer so when your string is longer than 32 it overwrites the begining of the command and the settings interpreter complains.

mcspr, 2018-01-27:
Oops, I've missed the wrapper.

Any reason it is 32 and not 128 as EmbedisWrap buffer?
https://github.com/xoseperez/espurna/blob/57bade36ed3adba439c7a1fdb5ccb384fe8bd98b/code/espurna/libs/EmbedisWrap.h#L13
```
aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa
-ERROR unknown command
aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa
-ERROR buffer overflow
```

xoseperez, 2018-01-27:
No reason except for memory usage. The EmbedisWrap value is the default value for Embedis too. I don't remember but I guess I just copied the signature. Do you have any good reason to have such long commands?

mcspr, 2018-01-27:
For me this breaks relatively long `set XXX` commands:
`set pass0 ...` with psk length 27 chars. (I guess this may be uncommon preference :)
`set mqttGroup0 hallway/group/lights` is 37 chars. This was set earlier using serial.

xoseperez, 2018-01-28:
Good points indeed. I'm changing it to 128.

xoseperez, 2018-01-28:
Pushed fix to dev branch. Will be released soon.

closed 2018-01-28 by xoseperez