xoseperez/espurna #478: Can't set mqttTopic to "actor/{identifier}" in telnet

byteborg, 2018-01-27:
Trying to set mqttTopic to a topic with prefix does not work (nothing happens). Setting it to `{identifier}` works. Am I doing it wrong?

```
set mqttTopic actor/{identifier}
set mqttTopic {identifier}
+OK
```

xoseperez, 2018-01-28:
Changing any setting related to MQTT topic in the terminal requires a reconnection to the broker. You just have to:

```
set mqttTopic actor/{identifier}
mqtt.reset
```

mcspr, 2018-01-28:
I think it is #477 
```python
>>> len("set mqttTopic actor/{identifier}\r\n")
34
```

Same blank response when exactly 32, and errors when it moves to the start of the buffer.

xoseperez, 2018-01-28:
In this case, the fix will be released with 1.12.2

closed 2018-01-28 by xoseperez