xoseperez/espurna #481: Fresh install, SPI missing in platformio.ini / lib_deps

PieBru, 2018-01-28:
Hi, after installing the fresh Arch linux on my new notebook I then installed PlatformIO and cloned Espurna from GitHub. Both pio 3.4.1 and 3.5.2a5 on both the master and dev branches gave build errors.

I solved by adding SPI in lib_deps and switching to 'platform = espressif8266' instead of 'platform = https://github.com/platformio/platform-espressif8266.git#v1.5.0'

Piero

xoseperez, 2018-01-28:
The error could be due to an unneeded reference to "spi_flash.h" in settings.ino. Can you try removing that reference and the SPI from lib_deps?

I'm building pre-compiled binaries against Arduino Core for ESP8266 2.3.0, hence the different platoform (v1.5.0).

PieBru, 2018-01-29:
Solved by cba621d, thanks.
Piero

closed 2018-01-29 by PieBru