xoseperez/espurna #483: Using two types of relay provider

cr3pt, 2018-01-29:
Hi!
I tried to use two types of relay for setting up LED dimming connected to ESP8266 -> NODEMCU_LOLIN

So I set

> #define RELAY_PROVIDER      RELAY_PROVIDER_LIGHT 
> #define LIGHT_PROVIDER      LIGHT_PROVIDER_DIMMER 
> #define DUMMY_RELAY_COUNT   2
> 
> // Buttons
> #define BUTTON1_PIN         0
> #define BUTTON1_MODE        BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH
> #define BUTTON1_RELAY       1
>
> #define BUTTON2_PIN         0
> #define BUTTON2_MODE        BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH
> #define BUTTON2_RELAY       1
> 
> // Relays
> #define RELAY2_PIN          12
> #define RELAY2_TYPE RELAY_TYPE_NORMAL
> 
> // Light
> #define LIGHT_CHANNELS      1
> #define LIGHT_CH1_PIN       13
> #define LIGHT_CH1_INVERSE 0

But I got only one switch button with dimmer, I didn't get any switch for RELAY2. Any solution?

xoseperez, 2018-01-29:
At the moment the firmware does not support mixing different relay types...

cr3pt, 2018-01-29:
So second question - is any solution for change PWM without LIGHT_PROVIDER_DIMMER?

xoseperez, 2018-01-29:
Currently, PWM in ESPurna is handled only by LIGHT_PROVIDER_DIMMER. You will probably need custom code for your requirements.

cr3pt, 2018-01-29:
OK :) I will try to change something inside ;)

etatus, 2018-11-07:
Same problem here with [GBLife RGBW Socket](https://github.com/xoseperez/espurna/issues/1305). If RGBW light is enabled, the socket cannot be turned on/off. If relay switch is enabled the RGBW light cannot be used.

Since RGBW can be turned on/off via brightness, a possible solution could be that      LIGHT_PROVIDER_DIMMER enable by default the light despite RELAY_PROVIDER_LIGHT was not specified. This way RGBW light could be used and the socket relay too.

etatus, 2018-11-11:
I've found a possible solution to make dummy and real relays work at the same time. Please check https://github.com/xoseperez/espurna/issues/1305#issuecomment-437672368

xoseperez, 2018-12-12:
Closing this as duplicate of the #1305

closed 2018-12-12 by xoseperez