xoseperez/espurna #484: Advices to build Espurna (custom version) with Atom and Platform.io ???

vincegre, 2018-01-29:
Hi

I'm trying to build custom version of Espurna for my specific ESP modules but I'm a "little" lost :(

I have Atom in my Linux system and installed in it Platform.io system and all needed dependencies.
I have downloaded a zip of Espurna github repo.
I open my local copy of the repo in Atom without problems.
In wiki it's stated to edit the arduino.h file to indicate what to compile but in mine everything is commented out. If I build project it still generates firmware for D1 mini realy shield !! Why ??

Also I guess I'll need to edit platform.io to add my hardware right ? and where do I decide which one to build ??

Thanks for help (pretty lost on this 👎 

Vincèn

cr3pt, 2018-01-29:
just change second line in code/platformio.ini to your, change config/arduino.h to your and that's all.

xoseperez, 2018-01-29:
@vincegre The arduino.h file is meant to easy build using Arduino IDE. If you are using PlatformIO you can still use it but it's recommended to use specific environments. You should see a list of options under the PlatformIO menu in Atom.

vincegre, 2018-01-30:
Thanks @cr3pt and @xoseperez for explanations and it looks easier to use Arduino IDE so I installed it following instructions in Espurna wiki ;) It looks like it compiles correctly the project if I try to.
Now I have well understood I select in arduino.h what I want to build (according to sensors, switches, etc I have) but how to do if I want to build my own firmware with custom config (let's say for example a D1 Mini with a CO2 sensor and temperature sensor wired on it). Does it mean I have to add a new hardware in program and if so how ? Should I add a new definition for my device in hardware.h ??

xoseperez, 2018-01-30:
You can add a new board definition to hardware.h, or, even better, create a "custom.h" file in the same folder and set your configuration there. You will then have to build the project with -DUSE_CUSTOM_H. Check the header in all.h file for more info.

That "custom.h" file could have:

```
// Info
#define MANUFACTURER        "WHATEVER"
#define DEVICE              "YOUWANT"

// Buttons
#define BUTTON1_PIN         0
#define BUTTON1_MODE        BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH
#define BUTTON1_RELAY       1

// Relays
#define RELAY1_PIN          12
#define RELAY1_TYPE         RELAY_TYPE_NORMAL

// SENSORS
#define DALLAS_SUPPORT      1
#define DALLAS_PIN          14
#define MHZ19_SUPPORT       0
#define MHZ19_RX_PIN        13
#define MHZ19_TX_PIN        15

```

MANUFACTURER and DEVICE are required, otherwise, it will complain.

vincegre, 2018-01-30:
Thanks @xoseperez a lot for the explanations so I'll go with create a custom.h for my own setup for each of my device ;)

I just have two issues:
-> if I try to compile project as downloaded from GitHub, I always get that error in Arduino IDE:

> In file included from /tmp/arduino_build_668470/sketch/config/all.h:17:0,
>                  from /home/vincen/vincen@toto.net/Protocoles-Docs/Espressif/ESPurna/Git Repository/code/espurna/espurna.ino:22:
> /tmp/arduino_build_668470/sketch/config/hardware.h:1432:6: error: #error "UNSUPPORTED HARDWARE!!"
>      #error "UNSUPPORTED HARDWARE!!"
>       ^

-> Trying to figure out how to pass the use custom argument at compiler :( It looks it has to be in platform.txt file if Arduino IDE so I added it at compiler.c.flags directive in that way: -D USE_CUSTOM_H but it doesn't look to have any effect or the previous error prevents it to work ??

Thanks a lot

Vincèn

xoseperez, 2018-01-30:
Yes, the Arduino IDE is not very friendly when you do something different. What I do is modify the boards.txt file adding extra_flags to the board definition I'm using. For instance, for the "Generic ESP8266 Module" that has "generic" as prefix I would add:

`generic.build.extra_flags=-DESP8266 -DUSE_CUSTOM_H`

vincegre, 2018-01-30:
@xoseperez thanks a lot for the trick so now it compiles fine and I'm able to generate the firmware file (name is a little weird but no matters: espurna.ino.generic.bin
I have flashed "my" firmware in my D1 but I encounter an issue. It starts well, I can connect to the AP but I'm stuck on password change webpage :( It keeps asking me for a new password, loads few seconds when I validate and comes back on that same page !!
I have then flashed the official firmware from github and no problem with that page !!
I noticed also something: title of webpages with official firmware displays Espurna followed by part of the MAC adress of the ESP module. With my firmware it displays Espurna 0.0.0

Not sure what to do now for that problem :(

xoseperez, 2018-01-30:
Do you have the chance to connect the device to your computer and open a serial monitor to it (baudrate 115200)? Check the debug log while booting the device and while trying to access that page. If you can, copy it here please.

vincegre, 2018-01-30:
good idea as in fact it looks like when I try to change passwd it reboots the ESP no ? logs below in minicom while authenticating with default passwd then trying to change it !

```
OPTIONS: I18n                                                                
Compilé le Feb  7 2016, 13:37:27.                                            
Port /dev/ttyUSB0, 14:01:27                                                  
                                                                             
Tapez CTRL-A Z pour voir l'aide concernant les touches spéciales             
                                                                             
[043458] [WEBSERVER] Request: GET /index.html                                
                                             [049678] [WEBSERVER] Request: GET /index.html
                                                                                          [059285] [WEBSERVER] Request: GET /index.html
                                                                                                                                       [060007] [MAIN] System OK
                                                                                                                                                                [065503] [WEBSERVER] Reql
                                                                                                                                                                                        l
                                                                                                                                                                                        s
                                                                                                                                                                                        a
                                                     
Exception (28):                                      
epc1=0x402210c0 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont 
          sp: 3fff2050 end: 3fff22e0 offset: 01a0

>>>stack>>>
           3fff21f0:  3fff2210 3fff0e88 3fff2250 40225980  
3fff2200:  00000007 3ffe8308 7fffffff 3fff6d2d  
3fff2210:  3ffe86e8 3fff6d2c 00000a00 4010020c  
3fff2220:  00000001 3fff227c 3fff2250 3fff12b4  
3fff2230:  3ffea720 00000000 000008a4 4010020c  
3fff2240:  00000001 00000001 3fff227c 3fff12b4  
3fff2250:  00000001 00000001 3fff0f14 4022123c  
3fff2260:  00000000 00000001 3fff0f14 4020816b  
3fff2270:  00000000 00000000 00000000 00000000  
3fff2280:  00000000 00000000 00000005 4020531c  
3fff2290:  3fffdad0 3fff0cf8 00000000 4020d7a0  
3fff22a0:  3fffdad0 3fff0cf8 00000004 40207c5c  
3fff22b0:  3fffdad0 3fff0cf8 0000000a 402032dc  
3fff22c0:  3fffdad0 00000000 3fff12ae 40223f68  
3fff22d0:  feefeffe feefeffe 3fff12c0 40100710  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v4ceabea9
~ld
   [000124] 

            [000125] [INIT] ESPURNA 1.12.3
                                          [000125] [INIT] xose.perez@gmail.com
                                                                              [000125] [INIT] http://tinkerman.cat

                                                                                                                  [000126] [INIT] CPU chip ID: 0xF1A1A2
                                                                                                                                                       [000127] [INIT] CPU frequency: 80z
                                                                                                                                                                                        )
                                                                                                                                                                                        0
                                                                                                                                                                                         
                                                                                                                                                                                         
                                                                                                                                                                                        F
                                                                                                                                                                                        z
                                                                                                                                                                                        T
                                                                                                                                                                                         
                                                                                                                                                                                        s
                                                                                                                                                                                        s
[... cut ...]
                                                                                                                                                                                        0
                                                                                                                                                                                        l

Side note: where is it supposed to generate the firmware file ? Right now I only find it in my tmp directory after compilation :(
```

vincegre, 2018-01-31:
So I have setup a dedicated Linux virtualbox instance for Arduino and everything needed for Espurna compilation and succeeded great by doing a custom.h and passing option at compile :)
I changed manufacturer name and product to see if it works fine as you can see in screen capture !
Will compile later version with sensors to check if it works fine for the autodiscovery and thanks again for all help and such great firmware Xose 👍 
[image]

closed 2018-01-31 by vincegre