xoseperez/espurna #486: Convert hPa to mm Hg

EvilSlesh, 2018-01-29:
Tell me how to convert hPa to mm Hg?
thx!

xoseperez, 2018-01-29:
google "hPa to mmHg"?

EvilSlesh, 2018-01-29:
I understand what should be divided into 133. )))
In the code ESPURNA in which place to add the conversion.

Sorry for my bad english

xoseperez, 2018-01-29:
You have to multiply by 0.75 (mmHg = 3 * hPa / 4). 

In sensors.ino there is a _magnitudeProcess meant to postprocess data from sensors, that includes unit conversion and resolution (# of decimals). Of course there should be some checking but for a quick hack that's the place.

EvilSlesh, 2018-01-29:
Thanks for the help.
Added in _magnitudeProcess 
```
if (type == MAGNITUDE_PRESSURE) {
        value = value * 0.75;  // 0.75 (mmHg = 3 * hPa / 4)
    }
```
web interface shows me 1010.10hPA, conversion dont work.
sorry i do not programmer, easy learning. )

xoseperez, 2018-01-29:
The units you will have to change them in html/custom.js and rebuild the web interface (check wiki). The value should have changed, was the flash OK?

EvilSlesh, 2018-01-29:
i stupid, works well. ;)
Another bin file, flashed.
Midnight i need to sleep, I'm tired.

thx!

xoseperez, 2018-01-29:
You are welcome

closed 2018-01-29 by xoseperez