xoseperez/espurna #487: telnet crash the system

kuppe234, 2018-01-29:
Hello,
with the new version 1.12.2 telnet crash the esp8266. I can connect, but with the first character the system crash. Serial-Debug is working.
=============================================================================
```
Exception (29):
epc1=0x40217643 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: sys 
sp: 3ffffca0 end: 3fffffb0 offset: 01a0

>>>stack>>>
3ffffe40:  3ffeb3ec 00000000 00000000 402484ff  
3ffffe50:  00000000 01b5b5cc 40102116 3fff6310  
3ffffe60:  00000000 00000002 3ffed3b2 40219824  
3ffffe70:  00000001 00007fff 00000000 4021a2e5  
3ffffe80:  00000001 3fff530c 3fff62bc 4021a308  
3ffffe90:  3fffc278 40101eb0 3fff62bc 402068c0  
3ffffea0:  3fff618c 00000000 4022b07f 00000001  
3ffffeb0:  3fff2420 3fff242c 00000006 3fff2060  
3ffffec0:  3fff618c 3fff2068 3fff2064 40206913  
3ffffed0:  3fff618c 3fff2068 3fff2064 4023e3b0  
3ffffee0:  00000002 3c01a8c0 00000018 00000002  
3ffffef0:  00000018 00000000 3fff0950 4010752c  
3fffff00:  00000000 4022e505 3ffeedd0 3fff2424  
3fffff10:  3ffed38a 3fff242c 3fff530c 4023b7a1  
3fffff20:  3fff1f80 3fff29bc 3fff29bc 3ffefbc0  
3fffff30:  00000000 3fff530c 0000001c 3fff29bc  
3fffff40:  3ffed37c 00000000 3fff530c 4023ab99  
3fffff50:  0901a8c0 00000064 00000000 00000035  
3fffff60:  00000002 0000001a 4022a9e3 3ffee5c0  
3fffff70:  3ffed354 3fffdcc0 3ffeabf0 3ffeabf0  
3fffff80:  4022a956 3ffee5c0 00000000 3fff2a1c  
3fffff90:  3fffdc80 00000000 3fff530c 402338ff  
3fffffa0:  40000f49 3fffdab0 3fffdab0 40000f49  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09826c6d
~ld
[000336] 

[000337] [INIT] ESPURNA 1.12.2
[000337] [INIT] xose.perez@gmail.com
[000337] [INIT] http://tinkerman.cat
```

=============================================================================
The version 1.12.1 is working correctly.
You need some more information?
Greetings, kuppe.

xoseperez, 2018-01-29:
Arg!
Stupid bug. Releasing again right now.

kuppe234, 2018-01-29:
Great! Version 1.12.3 works.
Many thanks, kuppe.

xoseperez, 2018-01-29:
Released as 1.12.3

closed 2018-01-29 by xoseperez

alexgavs, 2018-01-30:
Hello Xose
please check bool settingsRestoreJson(JsonObject& data) 
settings not restored from json.
there is a stack overflow, because the loop works in infinite mode and triggered watchdog.
should be> for (unsigned int i = EEPROM_DATA_END; i < SPI_FLASH_SEC_SIZE; **i--**)

xoseperez, 2018-01-30:
Can you check the value of SPI_FLASH_SEC_SIZE? I think it could be related to #481

Rapejim, 2018-02-06:
I have the same **Exception (29)** with 1.12.3:

`[124293] [MQTT] Sending NodeMCU/temperature/1 => 15.0 (PID 0)
[124294] [INFLUXDB] Sending to 192.168.1.107:8086
[124310] [INFLUXDB] Data: temperature,device=NodeMCU value=15.0
[124334] [MQTT] Sending NodeMCU/humidity => 24 (PID 0)
[124335] [INFLUXDB] Sending to 192.168.1.107:8086
[124348] [INFLUXDB] Data: humidity,device=NodeMCU value=24
[184326] [MQTT] Sending NodeMCU/temperature/0 => 28.6 (PID 0)
[184327] [INFLUXDB] Sending to 192.168.1.107:8086
[184345] [INFLUXDB] Data: temperature,device=NodeMCU value=28.6

Exception (29):
epc1=0x4022e888 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont 
sp: 3fff24c0 end: 3fff2ab0 offset: 01a0

>>>stack>>>
3fff2660:  000000f5 3fffb1d0 00000000 3fff2639  
3fff2670:  000000b0 3fffb208 0000002d 000011f5  
3fff2680:  00000011 3fff1994 3ffea090 3fffb1d4  
3fff2690:  0000008d 0000008d 3fff1358 4021b055  
3fff26a0:  3fff2750 3fff2720 00000008 401004d8  
3fff26b0:  00001f96 3ffe8308 00000100 0000008d  
3fff26c0:  3fff1358 00000008 3fffb1d4 40225544  
3fff26d0:  54534f50 72772f20 3f657469 683d6264  
3fff26e0:  5f656d6f 69737361 6e617473 3d752674  
3fff26f0:  696d6461 3d70266e 696d6461 5448206e  
3fff2700:  312f5054 0a0d312e 74736f48 0031203a  
3fff2710:  3fff2750 3fff2720 00000008 3ffea090  
3fff2720:  3ffea090 3fff2930 ffff8000 000000da  
3fff2730:  3fffbf50 00000000 00000cda 4010020c  
3fff2740:  3fff2780 0000008e 3fff2720 3fff2750  
3fff2750:  3fffb1d4 3fffb204 3fff2780 3ffea090  
3fff2760:  00001f96 3fffb0ec 3fff2880 4021170b  
3fff2770:  3fffb0ec 00001f96 00000025 3fff2880  
3fff2780:  54534f50 72772f20 3f657469 683d6264  
3fff2790:  5f656d6f 69737361 6e617473 3d752674  
3fff27a0:  696d6461 3d70266e 696d6461 5448206e  
3fff27b0:  312f5054 0a0d312e 74736f48 3931203a  
3fff27c0:  36312e32 2e312e38 3a373031 36383038  
3fff27d0:  6f430a0d 6e65746e 654c2d74 6874676e  
3fff27e0:  3733203a 0a0d0a0d 706d6574 74617265  
3fff27f0:  2c657275 69766564 4e3d6563 4d65646f  
3fff2800:  76205543 65756c61 2e38323d ffff0036  
3fff2810:  3ffe8308 00000000 0000000a 3ffeb3ec  
3fff2820:  3fff2893 3ffe960a 3fff2ec0 402362b8  
3fff2830:  3fff28a3 3ffe8a8a 3fffc200 3ffe9610  
3fff2840:  00000000 3ffe8308 3fff28f0 4022d6f0  
3fff2850:  00000000 ffffffff ffffffff 00000000  
3fff2860:  00000001 00000010 3f302064 00000035  
3fff2870:  3fff19bc 00000008 3fff2880 40225544 <
3fff2880:  706d6574 74617265 2c657275 69766564  
3fff2890:  4e3d6563 4d65646f 76205543 65756c61  
3fff28a0:  2e38323d 203e0036 00000000 49502820  
3fff28b0:  3fff2980 3fff2970 00000014 00000001  
3fff28c0:  3ffe960f 00000004 00000008 3fff2970  
3fff28d0:  3fff11ac 3ffe84e5 60000000 3fff1110  
3fff28e0:  3fff6ac4 3ffe8308 00000040 40229bfd  
3fff28f0:  3fff29b0 00000000 0000002f ffff0208  
3fff2900:  00000000 00000000 00000000 00000000  
3fff2910:  00000000 00000000 00000000 00000000  
3fff2920:  00000000 00000000 00000000 00000000  
3fff2930:  00000000 00000000 00000000 00000000  
3fff2940:  00000000 00000000 3fffb104 0000000f  
3fff2950:  0000000d 40218d68 3fff2980 3fff2970  
3fff2960:  3fffb0d4 3fffb1d4 3fffbf54 00000001  
3fff2970:  402185b4 3fff2a30 3fff2a30 3fff1110  
3fff2980:  3fff6ac4 3fff29f0 3fffb0d4 402117dd  
3fff2990:  3fffb0d4 3fff29f0 3fffb0d4 4021181c  
3fff29a0:  706d6574 74617265 2c657275 303d6469  
3fff29b0:  3fff2900 00001022 00001022 3fff1110  
3fff29c0:  3fff6ac4 4026ea3c 3fff2a30 4020495b  
3fff29d0:  706d6574 74617265 00657275 00000000  
3fff29e0:  3fff6ac4 00000000 00000001 40211c96  
3fff29f0:  362e3832 3fff2a00 40226d6c 40226b50  
3fff2a00:  3fffdad0 3fff0fbc 00000000 402125df  
3fff2a10:  4025278e 00000000 3fff134c 00000001  
3fff2a20:  40253917 3fff1560 3fff6314 40219e53  
3fff2a30:  3fffb0d4 0000000f 0000000b 4021f175  
3fff2a40:  00000000 00000000 00000000 00000001  
3fff2a50:  9999999a 403c9999 00000000 00000000  
3fff2a60:  9999999a 403c9999 3fff1110 3fff1a7c  
3fff2a70:  3fffdad0 3fff1464 0000000b 3fff1a7c  
3fff2a80:  3fffdad0 3fff1464 0000000c 40203e24  
3fff2a90:  3fffdad0 00000000 3fff1a76 40226db8  
3fff2aa0:  feefeffe feefeffe 3fff1a90 40100710  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v4ceabea9
~ld
[000106] 

[000107] [INIT] ESPURNA 1.12.3
[000107] [INIT] xose.perez@gmail.com
[000107] [INIT] http://tinkerman.cat`

I have two sensors installed:

`[000279] [GPIO] GPIO14 locked
[000295] [GPIO] GPIO12 locked
[000299] [SENSOR] Dallas @ GPIO14
[000299] [SENSOR]  -> temperature:0
[000299] [SENSOR] DHT11 @ GPIO12
[000299] [SENSOR]  -> temperature:1
[000300] [SENSOR]  -> humidity:0`

And if you need more info, say it me.

Un saludo.

darshkpatel, 2018-05-03:
@Rapejim Same Issue with resets and Exception (29)