xoseperez/espurna #488: Atom compiling error

Alveran81, 2018-01-29:
Need Help. I get this error.

[01/29/18 19:15:50] Processing magichome-led-controller (extra_scripts: extra_scripts.py; framework: arduino; lib_deps: ArduinoJson, https://github.com/marvinroger/async-mqtt-client#v0.8.1, Brzo I2C, https://bi
tbucket.org/xoseperez/debounceevent.git#2.0.1, Embedis, https://github.com/krosk93/espsoftwareserial#a770677, https://github.com/me-no-dev/ESPAsyncTCP#a57560d, https://github.com/me-no-dev/ESPAsyncWebServer#313
f337, https://bitbucket.org/xoseperez/fauxmoesp.git#2.4.2, https://bitbucket.org/xoseperez/hlw8012.git#1.1.0, https://github.com/markszabo/IRremoteESP8266#v2.2.0, https://bitbucket.org/xoseperez/justwifi.git#1.
1.6, https://github.com/madpilot/mDNSResolver#4cfcda1, https://github.com/xoseperez/my92xx#3.0.0, https://bitbucket.org/xoseperez/nofuss.git#0.2.5, https://github.com/xoseperez/NtpClient.git#b35e249, OneWire, P
MS Library, PubSubClient, https://github.com/xoseperez/RemoteSwitch-arduino-library.git, https://github.com/xoseperez/Time; build_flags: -g -DMQTT_MAX_PACKET_SIZE=400  -Wl,-Tesp8266.flash.1m0.ld -DMAGICHOME_LED
_CONTROLLER; lib_ignore: ; platform: https://github.com/platformio/platform-espressif8266.git#v1.5.0; board: esp01_1m; monitor_baud: 115200; board_flash_mode: dout)

 
Verbose mode can be enabled via `-v, --verbose` option
System: ESP8266 80MHz RAM/80K Flash/1M
UnicodeEncodeError: 'ascii' codec can't encode character u'\xe9' in position 151: ordinal not in range(128):
File "C:\Users\Goeran\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 158:
env.SConscript("$BUILD_SCRIPT")
File "C:\Users\Goeran\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 551:
return _SConscript(self.fs, *files, **subst_kw)
File "C:\Users\Goeran\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 256:
call_stack[-1].globals)
File "C:\Users\Goeran\.platformio\platforms\espressif8266@src-428e6fe6a638295277b17b20ae60f081\builder\main.py", line 375:
target_elf = env.BuildProgram()
File "C:\Users\Goeran\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
return self.method(*nargs, **kwargs)
File "C:\Users\Goeran\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 63:
env.BuildFrameworks(env.get("PIOFRAMEWORK"))
File "C:\Users\Goeran\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
return self.method(*nargs, **kwargs)
File "C:\Users\Goeran\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 261:
env.ConvertInoToCpp()
File "C:\Users\Goeran\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
return self.method(*nargs, **kwargs)
File "C:\Users\Goeran\.platformio\penv\lib\site-packages\platformio\builder\tools\piomisc.py", line 189:
out_file = c.convert(ino_nodes)
File "C:\Users\Goeran\.platformio\penv\lib\site-packages\platformio\builder\tools\piomisc.py", line 54:
return self.process(contents)
File "C:\Users\Goeran\.platformio\penv\lib\site-packages\platformio\builder\tools\piomisc.py", line 77:
assert self._gcc_preprocess(contents, out_file)
File "C:\Users\Goeran\.platformio\penv\lib\site-packages\platformio\builder\tools\piomisc.py", line 88:
fp.write(contents)
 [ERROR] Took 1.18 seconds

xoseperez, 2018-01-29:
This is a problem with PIO Core 3.5.2a4, check #457.

Alveran81, 2018-01-29:
Thank you very much.

closed 2018-01-29 by Alveran81