xoseperez/espurna #490: how to send rfin code via mqtt or http api to itead sonoff rf bridge?

prankousky, 2018-01-30:
hi,

i just flashed espurna to my itead sonoff rf bridge. works fine, i can connect to it, learn and save commands, etc.

however, i'd like to send rfin commands manually without teaching them to my bridge (mainly because i'd like to automate everything via mqtt and like to have the option of having more than 16 buttons)

here is what i did:

**learn** and **save** *switch #0*
switch on `26C0013603CA511451`
switch off `26D4013603D4511454`

sending these commands via the webgui **will** trigger my outlet, so the codes are correct. viewing the status via http API works as well.

mqtt displays the codes mentioned above via `dev-espurna-rf/rfin`. i tried **sending** these codes via mqtt, but no success. when i send the off command, mqtt will display it as received, but the rf bridge will not turn off my device.

i currently use node-red for home automation and would like to simply create a node that sends these on/off codes to my rf bridge **without** having to learn them first. i was going to figure out which device sends which code for on and off, then have *these* codes being transmitted via mqtt.

(since i try to switch as many outlets from 433mhz ones to smart outlets anyway, this *more than 16 devices* problem should only be theoretical; however, even if i could learn each command, i'd still prefer to send unlearned codes to the devices because it makes things easier)

is this possible, and if so, would somebody please tell me how i can achieve this?

thanks in advance :)

xoseperez, 2018-01-30:
rfin is for incomming messages (from the rfbridge point of view). You need rfout. So your topic would be "dev-espurna-rf/rfout/set" (remember to always add "/set" for commands). Just send the code as the payload, or additionally with the number of times you want it to be broadcasted:

`mosquitto_pub -t dev-espurna-rf/rfout/set -m "26C0013603CA511451,3"`

prankousky, 2018-01-30:
wow, that was quick!! thank you so much. this works perfectly. now i can finally get rid of my diy raspberry pi with 433 transmitter and only use the sonoff device. awesome!

xoseperez, 2018-01-30:
You are welcome 😄

closed 2018-01-30 by xoseperez