xoseperez/espurna #491: Interlock by software on Sonoff T1 2 gang, or Sonoff Dual

dote78, 2018-01-30:
Hi, i'd like to suggest an enhancement.

If there was a interlock mechanism embedded on Espurna for devices with 2 relays like Sonoff T1 or Sonoff Dual, it would be a great way to control motors with 2 directions like blinds or curtains (persianas). I don't know if it is difficult to do, something like "_when getting a signal to turn on a relay, first signal the other relay to turn off and wait some miliseconds (safety?), then turn on the asked relay_".

Thanks a lot Xose and the rest for this great firmware. Keep the good work!

xoseperez, 2018-01-30:
Except for the wait, the firmware already does that if you set the synchronization mode to "Zero or one switches active". In this sync mode, the firmware will check it never has more than one relay ON. One thing thou: right now it will action the relays in order, so there is a change to have both on for a few millis.

fmolero, 2018-02-02:
Forgive my English. How is this option activated?
Would there be a way to implement what the partner says? That is, when the order to activate a relay is given, first turn off all others, checking that they have been turned off and then turn on the relay that has received the command.

dote78, 2018-02-02:
The bit that Xose refers to is in the relay.ino file, relaySync function, lines 256-258 of the current build. These are the sequential changes he says.

for (unsigned short i=0; i<_relays.size(); i++) {
                if (i != id) relayStatus(i, false);
            }

And that is called in previous function called relayStatus. The loop seems ok to me, but I don't understand yet the previous lines to that to avoid the situation. Maybe it's as easy as adding some wait (of some miliseconds) after the loop completes to give all relays time to be effectively turned off. I can't see if the requested relay gets its target status and then relaysync is called, or the other way round.

I haven't had enough time to look it in depth and change it (and any related bit of code) to what I (and you) want, tho. It's the first time I look into the code of espurna. 

Let the weekend come! :)

xoseperez, 2018-02-02:
Those lines actually only enqueue the actions. This is part of the flow window strategy the firmware uses to avoid wearing out the relays too much too quickly. The actual relay switching is done in the `_relayProviderStatus` method that is called from the `relayLoop` method.

https://github.com/xoseperez/espurna/blob/f8907d60395ccfe0d606b4fb24624720fbcd7f51/code/espurna/relay.ino#L773

And that's where it loops the relays in order. That loop should be done twice. The first time to turn off those relays that must be turned off and the second one to turn on the relays that must be turned on.

xoseperez, 2018-02-02:
I have pushed a commit that forces turning off relays before turning others on, basically calling the loop in relayLoop twice with different target modes.

dote78, 2018-02-02:
Thanks a lot Xose. Keep up the good work!!!

fmolero, 2018-02-02:
Thanks for the modification. 

In this case, if I set the mode "0 or one switch active" it would serve so that when you activate, for example, lowering a shutter, before it stops rising in case it is activated, right?

dote78, 2018-02-02:
That is what I understand, yes

fmolero, 2018-02-02:
I have two sonoff basic to control a blind, it's possible use "One and just one switch active" or i need use sonoff dual?

dote78, 2018-02-03:
A firmware on a device will only control the relays on its device, so you need a device with 2 or more relays for this option to be of any use for you. Dual, 4ch or T1 for instance.
For syncing between devices you have the option to set up MQTT groups (in the "Switches" tab) where you can set the group state to "Inverse", but you have no guarantee of the order it will happen, as each device will get the notification in its own way, so not useful to control a motor with 2 directions as you will probably get simultaneous current at some point.

xoseperez, 2018-02-03:
@fmolero  The dual has only one external button and two more in a header so you can add your own button. Mind there are two versions of the dual and they differ a lot (original and R2). 

@dote78  is right about the MQTT group feature and its cons.

fmolero, 2018-02-04:
Thanks, I've asked for one to prove that I think it's version r2

xoseperez, 2018-02-26:
This is staged for release.

closed 2018-02-26 by xoseperez