xoseperez/espurna #492: Question about dummy relay

brahmafear, 2018-01-30:
Looking to use an esp8266-01 module with DHT11/22 connected on gpio2.  I've got this working on the module by configuring it for a non-existent relay on gpio0, but I'd like to utilize the dummy relay config (used by the Espurna compatible lights) to remove the unnecessary switch in the web config.  I don't see any documentation about DUMMY_RELAY_COUNT so was wondering if you could provide the correct config directives to add to hardware.h?  I can then submit a PR if you are interested in including in master branch.

xoseperez, 2018-01-31:
Sorry, I don't get how the DHT is related to the relay on GPIO0.

brahmafear, 2018-01-31:
Apologies for lack of clarity.

I am hoping to use Espurna on an esp8266-01 module with a DHT11 or DHT22
connected to gpio2.  I am inquiring about the best config directives in
hardware.h to represent this.

Thanks.

xoseperez, 2018-01-31:
Just enable DHT support in the sensors.h file (search for DHT_SUPPORT) and configure the right GPIO in the DHT_PIN setting.

https://github.com/xoseperez/espurna/blob/f8907d60395ccfe0d606b4fb24624720fbcd7f51/code/espurna/config/sensors.h#L179-L185

brahmafear, 2018-01-31:
> 20fbcd7f51/code/espurna/config/sensors.h#L179-L185

Already have that in place.  I'm wondering if it is possible to disable the
switch/relay config.  Ie, number of relays == 0.

Thanks.

xoseperez, 2018-02-01:
In the hardware.h file look for the device you are building for (Sonoff Basic?) and remove the RELAY configuration parts.