xoseperez/espurna #495: No Response to MQTT topic

tarbax, 2018-01-31:
can anyone give the list of mqtt topics from espurna?

i am trying to get it to work without succes.
relay?
    state_topic: '/Hostname/Client ID/relay/0'
    command_topic: '/Hostname/Client ID/relay/0/set'
brightness?
    brightness_command_topic:  '/Hostname/Client ID/brightness/set'
    brightness_state_topic: '/Hostname/Client ID/brightness'
my board: quinled
Thanks!

xoseperez, 2018-02-01:
I'm working on documenting the MQTT topics. PLease check the [MQTT](wiki/MQTT) page in the wiki (still a work in progress).

tarbax, 2018-02-01:
i was doing it right something else is wrong with it.
it dedects and connects with the mqtt broker
settings:
GENERAL
Hostname: quinled01
MQTT
MQTT Retain: enabled
MQTT Root Topic: {hostname}
Use JSON payload: enabled

Home assistant:
  - platform: mqtt
    name: 'quinled keuken'
    state_topic: '/quinled01/relay/0'
    command_topic: '/quinled01/relay/0/set'
    payload_on: 1
    payload_off: 0
    brightness: true
    brightness_scale: 255
    brightness_command_topic: '/quinled01/brightness/set'
    brightness_state_topic: '/quinled01/brightness'

xoseperez, 2018-02-01:
If your root topic is` {hostname}` the relay state topic will be `quinled01/relay/0` and the command `quinled01/relay/0/set`.

tarbax, 2018-02-01:
@xoseperez 
so why is it that it is not responding ? what could the issue be ?
other mqtt devices work without problems

xoseperez, 2018-02-01:
Please, read my previous message. Disable "Use JSON topic" and remove the leading slashes in your HA configuration.

tarbax, 2018-02-03:
Sorry had a old config copied this is the current everything works only my relay state is not working  
- platform: mqtt
    name: 'quinled'
    state_topic: 'quinled01/relay/0'
    command_topic: 'quinled01/relay/0/set'
    payload_on: 1
    payload_off: 0
    brightness: true
    brightness_scale: 255
    brightness_command_topic: 'quinled01/channel/0/set'
    brightness_state_topic: 'quinled01/channel/0'

xoseperez, 2018-02-03:
The QuinLED is an RGB module, right? You should set the brightness state and command to `quinled01/brightness` and `quinled01/brightness/set`. Do you have Use JSON topic enabled?

tarbax, 2018-02-03:
@xoseperez no it is a 2 channel led dimmer nothing rgb and yes json is enabled /brightness/set was also not working

xoseperez, 2018-02-03:
Turn off "Use JSON payload".

closed 2018-08-20 by stale[bot]