xoseperez/espurna #496: Question about adding new hardware

sunbit, 2018-01-31:
Hi, in the last days i succeded in using espurna in a cheap esp-01 relay board from aliexpress, and I was wondering if it's worth contributing that to the documentation. The point is that that board needs a simple modification, without it, the board doesn't boot in the right way, and that's the interesting thing more that the board itself.

I used the 'jorgegarcia relays' env for testing it, selected it while i was looking for one using esp-01. In case you consider that this is worth adding, should i add a new env for this specific board?

Thank you

xoseperez, 2018-01-31:
Sure thing, the proper way is to issue a pull request. The minimum is to add the board definition to the hardware.h file. Additionally you can add an environment to the platformio.ini file (this way a binary file will be generated with each release) and also add a reference to it in the arduino.h file.

nkerezov, 2018-02-03:
Yes guys it would be super cool to have binary file for a simple esp 8266-01(GPIO 0and2),without waithing for v.2.0