xoseperez/espurna #497: Feature request: Sonoff S31 support

pjz, 2018-01-31:
Any chance that [S31 (Smart plug with energy monitoring)](https://www.itead.cc/smart-home/sonoff-s31.html) support is forthcoming?

xoseperez, 2018-01-31:
According to this source (http://www.phreakmonkey.com/2018/01/sonoff-s31-disassemble-and-flash.html) the button and relay GPIOs are the same than in the Sonoff Basic. 

Then there is a ChipSea 7766 energy monitoring chip ([datasheet](http://www.chipsea.com/UploadFiles/2017/08/11144342F01B5662.pdf)) connected via UART.

I have requested for a sample from Itead to try it out.

swilson86, 2018-03-28:
The Sonoff S31 is a very interesting smart plug as it doesn't have all the wires dangling out like the POW. Perfect for a coffee pot or fridge where nearby liquids can be present. Anxious to see some Espurna development on this too.

xoseperez, 2018-04-04:
Working on this in the sensors branch. Still untested. 0e6f2d1ca39e28a8fbad6481fc7498c4a0c5814f

xoseperez, 2018-04-16:
It is now merged into the dev branch and staged for release.

xoseperez, 2018-04-18:
Staged for release. Closing.

closed 2018-04-18 by xoseperez