xoseperez/espurna #499: HLW8012 with tied SEL pin

pjz, 2018-02-01:
Hi, I'm working on putting Espurna onto a board that has a HLW8012 with the SEL pin tied HIGH so it will only read voltage.  How can I configure Espurna for this kind of setup?

pjz, 2018-02-01:
I can, of course, make something up (probably something like #define some HLW8012_SEL_PULLUP/PULLDOWN constants and then make HLW8012Sensor.h configure _hlw8012 appropriately), but since you're the author of both libraries I thought it would be good to see if you had any ideas or if there was some functionality somewhere that I'd overlooked.

xoseperez, 2018-02-01:
Actually, there is nothing in the HLW8012 to achieve this so you will need to modify both the hlw8012 library and ESPurna. I would change the `toggleMode` method in the library to stay at `MODE_VOLTAGE` and remove the `MAGNITUDE_CURRENT` slot in ESPurna.