xoseperez/espurna #502: Scheduler : update/enable via MQTT

bopnrgi, 2018-02-01:
Scheduler is great, but it have to be set from the Espurna Web Interface only, that's fast and Nice.

It could be possible to enable/disable it (and set parameters like hours/minutes and relay action) from an MQQT message ?

Scheduler could be set by home assistant (or other), I prefer use espurna's scheduler (reliable and standalone) but in case of loss of wireless connection the espurna => espurna doing the job :)

xoseperez, 2018-02-01:
Mmm... I don't think this is necessary. If you already have the logic somewhere else there is no need to send the scheduler to the device via MQTT, you can just turn it on/off from HA or Node-RED or whatever using MQTT.

bopnrgi, 2018-02-01:
This suppose that the connection between HA and Espurna is permanent ?
If no, maybe i miss something with your explanation ?

The scheduler logic could be coded in HA (and then HA switch ON/OFF the relay), it's true. 
But, if HA is not available or broken connection (at the expected switching time) the relay can't be toggle (except with espurna scheduler). i'm true ?

xoseperez, 2018-02-01:
The connection between HA and ESPurna is via MQTT. Note that the scheduler also relies on having internet connection at least once to sync the clock via NTP.

closed 2024-08-30 by mcspr