xoseperez/espurna #503: HC-SR501 PIR MOTION DETECTOR

MariusLaurinaitis, 2018-02-01:
Hello Xose,
You are growing fast! It would be very cool to add PIR sensor (HC-SR501) support too like [https://www.mpja.com/download/31227sc.pdf] It would be possible to see status through mqtt. Then one Wemos D1 or NodeMCU could work with a whole set of sensors.

And if we could even plug an OLED screen (popular monochrome 128x64, SPI SD1306 driver or SPI SH1106 driver) to output mqtt values :) 

Best regards
Marius

gretel, 2018-02-01:
@MariusLaurinaitis have four `sonoff basic` running with `HC-SR501` connected to `GPIO`.

the according part of  `hardware.h` looks like this:

```
    #define BUTTON2_PIN         14
    #define BUTTON2_RELAY       0
    #define BUTTON2_MODE        BUTTON_PUSHBUTTON
    #define BUTTON2_PRESS       BUTTON_MODE_ON
    #define BUTTON2_CLICK       BUTTON_MODE_OFF
    #define BUTTON2_DBLCLICK    BUTTON_MODE_NONE
    #define BUTTON2_LNGCLICK    BUTTON_MODE_NONE
    #define BUTTON2_LNGLNGCLICK BUTTON_MODE_NONE
```

works fine and gets published via `mqtt`.

@xoseperez btw - i'd love to have a way to get this discovered by `home-assistant`

MariusLaurinaitis, 2018-02-01:
@gretel thank you very much!

xoseperez, 2018-02-01:
You have the digital sensor that does just that: report the status of a digital GPIO.

cookier0218, 2018-03-29:
Hi Xose,
I wanna use HC-SR501 connected to GPIO 3 of sonoff basic. The part of setting hardware.h is
    
    #define MANUFACTURER        "ITEAD"
    #define DEVICE              "SONOFF_BASIC"

    // Buttons
    #define BUTTON1_PIN         0
    #define BUTTON1_MODE        BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH
    #define BUTTON1_RELAY       1
    #define BUTTON2_PIN         14
    #define BUTTON2_MODE        BUTTON_SWITCH | BUTTON_SET_PULLUP | BUTTON_DEFAULT_HIGH
    #define BUTTON2_RELAY       1
    #define BUTTON3_PIN         3
    #define BUTTON3_MODE        BUTTON_SWITCH | BUTTON_SET_PULLUP | BUTTON_DEFAULT_HIGH
    #define BUTTON3_RELAY       2
    // Relays
    #define RELAY1_PIN          12
    #define RELAY1_TYPE         RELAY_TYPE_NORMAL
    #define RELAY2_PIN          4
    #define RELAY2_TYPE         RELAY_TYPE_NORMAL
But it can't work, is there any other setting or problem with my settings?
regards,
cookier0218

xoseperez, 2018-04-04:
GPIO3 is RX in the Sonoff Basic, you are probably producing noise there. Why don't you use GPIO14 which is in the header?

xoseperez, 2018-07-02:
Closing this since there has been no feedback in 3 months. Feel free to reopen if necessary.

closed 2018-07-02 by xoseperez

nqdanyb, 2018-07-12:
As I think, it is possible to add a PIR sensor (HC-SR501) as a sensor like the DHT sensor in arduino.h so that ESPURNA can display the state of the PIR sensor on the WEB interface and publish. via MQTT. I tried attaching a PIR sensor to the GPIO pin that supports DIGITAL_SUPPORT and enabling the "#define DIGITAL_SUPPORT" sensor in the SONOFF_BASIC platform. Then the data from the PIR sensor is sent to the home-assistant, which then activates the light when the sensor senses motion. But the high latency caused by the time from detecting motion to sending home-assistant information took more than 6 seconds. What can I do to reduce this latency?
Here are some screenshots:
[image]
[image]

nqdanyb, 2018-07-17:
Thanks @xoseperez. I tried the latest version of the espurna-sensors, but the delivery time of the PIR status (HC-SR501) from detecting motion to the Home Assistant through the MQTT is still very slow. I then modified the sensors.h file as shown below:
[image]
After correction, the time to send the PIR status (HC-SR501) from the motion detector to the Home Assistant through the MQTT has been improved (about 1 second)

gretel, 2018-07-17:
@nqdanyb  my ["old" approach](https://github.com/xoseperez/espurna/issues/503#issuecomment-362274817) works without any noticeable latency.

nqdanyb, 2018-07-18:
Thanks @gretel . I have edited the file hardware.h as your guide. It is true that the PIR state (HC-SR501) reads very fast. But I do not know how to send it via MQTT to Home Assistant
[image]

xoseperez, 2018-07-26:
There is no support to send button event to HA at the moment.
You have to configure the events in the EventSensor config section setting EVENTS_TRIGGER to 1.