xoseperez/espurna #504: feature request: Deep Sleep

pascal1404, 2018-02-01:
Hi, first of all you do a great job with Espurna.
I have a ESP6288-12E with a DHT22. The ESP is battery powered so I would like to use Espurna with deep sleep mode. Is this possible? I would like to send the data via mqtt every 10min. If I have to change the settings of Espurna, I would like to publish a value to the mqtt-server with retain-flag. If Espurna go online after deep sleep, it gets this value and it should be remain online to change settings. After a reboot it should be go in deep sleep mode again. 

Also is it possible to use the hostname of the mqtt-server instead of his IP-adress?

Thanks Pascal

xoseperez, 2018-02-01:
Thanks! 
I'm taking note of the suggestion.

You can use hostnames in the MQTT server setting. If the hostname is a mDNS hostname (.local) you have to enable MDNS_CLIENT_SUPPORT.

kevindgoff, 2018-02-05:
I also thought this would be helpful for outside sensors/monitors that don't have the luxury of power nearby.  I made a tilt sensor for my garage that was originally powered by battery.  Eventually, I ended up wiring it to power because the battery life was unacceptable.

axwax, 2018-05-03:
+1
I'd love to be able run Espurna on my battery-powered ESP in the garden too, rather than having to use my custom firmware...

hyteoo, 2018-05-19:
+1 on this
I love espurna too much not to use it for my weather station.
It would be great if you could gradually introduce some power management. 
Also maybe some battery charge status support in the future.

pyrosmiley, 2018-06-27:
Put me down for a big +1 here. 

Deep sleep would be absolutely _massive_ for me. I also hadn't thought of battery charge status but that's a good one!

mitchmitchell, 2019-03-13:
I would like to be able to use espurna on battery powered devices with deep sleep support as well.

pascal1404, 2019-03-13:
I found a solution to use deep sleep. I use now esphome from esphome.io. It is a great framework too. The only disadvantage is the ugly websever.