xoseperez/espurna #505: feature request: extend MQTT json payload feature

hyteoo, 2018-02-01:
Currently there is a switch which enables all MQTT messages to be sent as a JSON string (except status)
However this is not very convenient when having more than one output (2ch, 4ch 8ch, RGB, etc)

Solution:
add switches enabling individual parts to be sent as JSON:
- status
- relay/output status
- heartbeat

xoseperez, 2018-02-01:
Can you clarify (with an example maybe) what would be your desired payloads for those types?

serhatozkara, 2018-02-01:
+1 Home Assistant is a good example. MQTT JSON Light component supports more features. I think it would be better to advertise to home assistant  as mqtt json light when JSOn is selected under MQTT settings.

hyteoo, 2018-02-01:
Ok let's break down the messages espurna sends out:

- individual relay status for each channel
- other values like brightness, temperture, humidity, etc
- heartbeat
- status

my point is, add some more switches in the MQTT page that enable each group JSON output.

My original problem was that for example, a Sonoff basic or S20 - it has 1 relay, so that would imply low MQTT traffic
But heartbeat sends out a lot of info over several MQTT messages.
It would be more elegant to send the heartbeat only as JSON, keep the relay/0 value in a non JSON message for easy parsing (much easier to read the raw payload as 1 or 0, instead of parsing a JSON with 1 element.

jonny190, 2018-02-21:
+1 for home assistant MQTT_Json support (https://home-assistant.io/components/light.mqtt_json/)