xoseperez/espurna #506: Support Feican LEDs

serhatozkara, 2018-02-01:
I successfully flashed two feican RGBW light bulbs with Espurna. It would be great to add support for these lights as they are ESP8266 based cheap RGBW lightbulbs. I modified LC02 hardware pin definitions as following: R: 12 G:13 B:14 W:5
But after I compile, i can't access the web management interface from Safari, I get websocket 401 error. I can upload flashing instructions if needed.

gn0st1c, 2018-02-01:
@serhatozkara : which feican? rgbw or rgbw02?
(iclerini zarar vermeden acmak kolay mi? bazilari epoxy dolu oluyor)

xoseperez, 2018-02-01:
@serhatozkara  Are you using the latest version from the repo?

serhatozkara, 2018-02-02:
Yes, I'm using the master. I'm also using esp8266 core 2.4.0 and Arduino IDE 1.8.5
@gn0st1c you can seperate it with a razor i'll post instructions and the bin to my github.

serhatozkara, 2018-02-02:
@xoseperez OK, i've recompiled project files with Platform IO which includes the libraries you defined on platform IO ini. Theres is no Safari bug when i used that libraries. However, something strange happened:
I uploaded the firmware Platform IO compiled and one of the bulbs upgraded beautifully and the other one stuck in a reboot loop (Blue LED constantly on). Then I tried uploading the same image with FTDI cable, to my surprise, that bulb stuck in a reboot loop again. I recompiled firmware image using arduino ide with the libraries Platform IO pulled, this time it worked. There is something I miss here definitely. 
Also, firmware bin file from Arduino IDE is 517 kb one from platform IO is 487 kb. BTW, on Arduino IDE compiled firmware, it keeps connecting/ disconnecting to TCP and MQTT , which renders web interface and telnet unresponsive.

xoseperez, 2018-02-03:
The default PlatformIO environments are compiled against Arduino Core 2.3.0 while you might have the latest 2.4.0 version on your Arduino IDE. Of course, there are a lot of differences between them, but one of these differences is that the image generated using 2.4.0 is ~40Kb bigger.

Also, the version of ESPAsyncWebServer linked from PIO is not the latest one, since there is an issue with WebSocket authentication (!!) on the latest versions.

serhatozkara, 2018-02-10:
Hello @xoseperez, I've been using the lamps for one week now mostly without issues. However one issue is critical for me: For the warm white channel, sometimes lights go full brightness for a split second then reboots. The other issue is, i can never set the warm white channel to full brightness cause after some 20 seconds or so, lights start flickering badly. I can only turn it up to 170/255 without this behaviour. Only warm white channel does this flickering issue. Seems like a PWM issue. Can you point me in the right direction on this? Also one of the bulbs keeps bricking with OTA :)

xoseperez, 2018-02-11:
PWM support is done using Steffan Bruens' ESP8266_new_pwm code (https://github.com/StefanBruens/ESP8266_new_pwm). One of the most common issues when this happens is power supply, have you checked this?

serhatozkara, 2018-02-18:
Hello @xoseperez 
I suspected power supply might be a problem. But units seem to be working fine with stock firmware. Of course they might be limiting the output to some value as on stock firmware you can't turn on warm white with color leds at the same time. 
I have a few questions here:
I tried modifying LIGHT_LIMIT_PWM to 5000, the light output is the same. Although the light seems to be more responsive to value changes and generally more smooth. Can you clearify how this variable affects the lights and is this the correct way to limit the power?
Can I utilize SDK_PWM_PERIOD_COMPAT_MODE to 1 in order to improve compatibility with this light?

Although this is a stretch, I might be setting the wrong pin on warm white channel which i set to 5,
may be i'm using the wrong channel for warm white but somehow  this pin somehow also control the warm white channel. Can this happen?

I mentioned one of the bulbs is not accepting any firmware from platform.IO. Well it turns out it is not about platform.IO but the esp core. I compiled a the firmware with arduino ide using Core 2.4.0. When i switch back to core 2.3.0 on arduino ide, it doesn't accept the firmware also. Somehow it only accepts firmwares compiled with core 2.4.0 and I can't seem to switch back to 2.3.0. Did you encounter such a behaviour?

I also noticed a strange behaviour with MQTT. When the firmware can't communicate with the MQTT server defined in the settings (mosquitto server down or other problems), firmware becomes very unstable and Lights start to flicker heavily almost like a feedback loop and eventually turn off(most probably an exception). At that point, the unit never recovers and have to powercycle to communicate with the bulb again. Sometimes even all settings are reset.

**Off topic: New Philips Xiaomi bulbs are using ESP8266! They're cheap and really good.