xoseperez/espurna #507: Web Interface not working on Safari/iOS

xoseperez, 2018-02-02:
As reported by @serhatozkara.

The web management interface does not work from Safari. It returns a websocket 401 error. 

This was allegedly solved a few week ago. Check #295. But a number of comments in issues #295, #506 and #384 suggest the opposite.

I cannot test this myself, I will appreciate help from anyone willing to test/hack the web.ino and websockets.ino modules.

Cabalist, 2018-02-02:
I've previously posted the things I see in a browser on the other thread.  Is there a serial terminal I can watch for debugging output?  

Would a remote session be of any value?   I could provide you with a macOS with a fresh install and a couple of espurna devices on a network.

xoseperez, 2018-02-03:
If building the firmware yourself, please double check you have the proper version of ESPAsyncWebServer. There is an issue with WebSocket authentication (!!) on the latest versions. The recommended version is me-no-dev/ESPAsyncWebServer@313f337.

chryss44, 2018-02-09:
1.12.1 work fine with SAFARI on Apple but not 1.12.3 :-(.

thebradleysanders, 2018-02-19:
The current version of Espurna seems to open fine on an Ipad in Safari, but not on Iphone.

tripplet, 2018-03-13:
I have the same problem with safari 11.0.3 (13604.5.6) on macOS High Sierra with the official ESPURNA v1.12.4.
It looks like the problem is the authentication of the web socket connection.

Safari reports: [Error] WebSocket connection to 'ws://device.local/ws' failed: Unexpected response code: 401

This seems to be related to https://bugs.webkit.org/show_bug.cgi?id=80362.
Also when I disable "WS Auth" in the web ui safari works again.
Maybe there is a workaround to use another way of authentication for the websocket connection until this is properly fixed in safari.
Based on the bug history I would not count on a fix by apple.

ber-red, 2018-03-29:
same Problem here:
A) Safari 11.0.3 (12604.5.6.1.1) on macOS Sierra with the official ESPURNA v1.12.4.
After entering the credentials I get these 2 Errors:
1) `[Error] Failed to load resource: the server responded with a status of 404 (Not Found) (favicon.ico, line 0)`
2) `[Error] WebSocket connection to 'ws://<host ip address>/ws' failed: Unexpected response code: 401`
and a blank page
B) Safari version ?? on iOS 11.2.6 (15D100)
After entering the credentials I get a blank page (not sure if i could to do any debug on my iPhone)

xoseperez, 2018-04-04:
Please check #667 .

closed 2018-04-04 by xoseperez