xoseperez/espurna #508: feature request: Dimmer with brightness local control

hexrus, 2018-02-02:
Hello, I want attach button to GPIO04 MagicHome controller and control brightness with it button.
Button click  - on/off light
Button hold  - cyclic brightness dimmer
I can`t find this functions in espurna firmware. It`s realy?

xoseperez, 2018-02-02:
You can define a button for the MagicHome in hardware.h that will do the on/off. But the dimmer part is not available.

In the MagicHome section of the hardware.h file add these lines. Connect the pushbutton between GPIO4 and ground.

```
    // Buttons
    #define BUTTON1_PIN            4
    #define BUTTON1_MODE        BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH | BUTTON_SET_PULLUP
    #define BUTTON1_RELAY       1

```

hexrus, 2018-02-02:
Yes, I already test this solution. I intresting do you planning add button dimmer feature.

xoseperez, 2018-02-03:
No, I did not have that in mind... sorry. You can try adding it yourself and do a pull request.