xoseperez/espurna #509: Factory reset does not work? (Duo)

github-0, 2018-02-02:
I installed espurna to Sonoff Duo few weeks ago (sorry, don't know the exact version anymore) and probably changed the wifi AP password since I cannot connect to it anymore. SSID shows up no problem and I can switch the relay manually from the button. I have tried to hold the button for more than 10 seconds but nothing seems to happen and still cannot connect to wifi. Similarly long click (1 sec) doesn't seem to reboot the device. Should there be an indication that these actions are completed? What should I do in this situation to regain access?

xoseperez, 2018-02-03:
The original Dual (not the R2 version) does not have the factory reset option since the button is handled by a secondary microcontroller, not the ESP8266. Your only option is to open it again, do a flash erase and flash the firmware again.

Check the instructions in the [Binaries](https://github.com/xoseperez/espurna/wiki/Binaries) wiki page.

github-0, 2018-02-04:
Thank you. Wish I had kept better track of the password but knowing this I most certainly will from now on.

closed 2018-02-04 by github-0