xoseperez/espurna #512: Magic Home LED Controler Board v2.3

Alveran81, 2018-02-02:
I have Magic Home LED Controler Board v2.3 with RGB (without white channel)

Working configuration is:
  // Light
    #define LIGHT_CHANNELS      4
    #define LIGHT_CH1_PIN       5      // RED
    #define LIGHT_CH2_PIN       12       // GREEN
    #define LIGHT_CH3_PIN       13      // BLUE
    #define LIGHT_CH4_PIN       15      // WHITE

I thing this should also work
  // Light
    #define LIGHT_CHANNELS      3
    #define LIGHT_CH1_PIN       5      // RED
    #define LIGHT_CH2_PIN       12       // GREEN
    #define LIGHT_CH3_PIN       13      // BLUE

xoseperez, 2018-02-03:
Looks like its exactly the same as v2.0. You are not the only one to report this. I guess the original info I had was wrong. I'm removing the specific configuration for this board. Use 2.0 instead.

xoseperez, 2018-02-26:
Staged for release

closed 2018-02-26 by xoseperez