xoseperez/espurna #513: HLW 8012 Wrong Readings

varungujjar, 2018-02-02:
Hey @xoseperez 

so basically i ordered for a HLW8012 breakout board from spain.. and have been testing current along with it.(10W, 60W & 100W Bulbs ). however after calibration and resets.. I still get wrong readings.. am I doing something wrong.. ?

So I calibrated as per a 100 W bulb with 225 volts( using multimeter )
100W/225V = 0.444
when I add this calibration value and switch on the switch  it gives me right values however when I replace it with my 60W & 10 W bulb my readings are completely off here are my readings

0.44(i)  225(v)  100(w) for a 100W bulb ( works fine with a 100W bulb )
0.35(i)  225(v)  82(w)  for a 60W Bulb
0.26(i)  225(v)  59(w)  for a 10W Bulb

varungujjar, 2018-02-03:
Hey @xoseperez 
figured it out .. the breakout board 
http://www.electrodragon.com/product/energy-meter-hlw8012-breakout-board/

has a 0 ohm resistor missing from the Live phase to the GND... works fine now.. :)

has  a

xoseperez, 2018-02-03:
👌

closed 2018-02-03 by xoseperez

varungujjar, 2018-02-06:
@xoseperez since the HLW 8012 either supports voltage readings or current ( while measuring current the voltages are always wrong but my current readings are accurate )  how can I hide other magnitudes on the web interface.. and make sure only active power is seen..