xoseperez/espurna #514: WEMOS D1 mini and V9261F and influxdb, latest dev, crash after 4h

fork04, 2018-02-03:
crash
[161737] [DEBUG] Crash at 808 ms
[161741] [DEBUG] Reason of restart: 2
[161741] [DEBUG] Exception cause: 0
[161742] [DEBUG] epc1=0x40202f50 epc2=0x00000000 epc3=0x00000000
[161744] [DEBUG] excvaddr=0x00000000 depc=0x00000000
[161748] [DEBUG] >>>stack>>>
[DEBUG] 3fff1950: feefeffe feefeffe feefeffe feefeffe
[DEBUG] 3fff1960: 2c09f2b1 feefeffe feefeffe feefeffe
[DEBUG] 3fff1970: feefeffe feefeffe feefeffe 00000001
[DEBUG] 3fff1980: 20037025 00000002 00000000 40107635
[DEBUG] 3fff1990: 401070d0 feefeffe feefeffe 4010711e
[DEBUG] 3fff19a0: ffffffff 00000020 feefeffe feefeffe
[DEBUG] 3fff19b0: feefeffe feefeffe feefeffe 00000022
[DEBUG] 3fff19c0: 3fffc200 40107098 3fffc258 4000050c
[DEBUG] 3fff19d0: 400043e6 00000030 00000016 ffffffff
[DEBUG] 3fff19e0: 400044ab 3fffc718 3fff1ad0 08000000
[DEBUG] 3fff19f0: 60000200 08000000 08000000 00000000
[DEBUG] 3fff1a00: 0000ffff 00000121 ffffffff 003ff000
[DEBUG] 3fff1a10: 402686f0 0000001c 003ff000 00000030
[DEBUG] 3fff1a20: 3fff1a64 00000005 00000a50 00000000
[DEBUG] 3fff1a30: 00000a41 00000a5d 00000a40 3ffe8924
[DEBUG] 3fff1a40: 00000010 3fff1aec 3fff1aec 40202188
[DEBUG] 3fff1a50: 3f323931 00000001 3fff1aec 4020a41e
[DEBUG] 3fff1a60: 00000002 00000000 00000001 00000006
[DEBUG] 3fff1a70: 4022f52e 3fff1b61 3fff5ba9 3fff1b10
[DEBUG] 3fff1a80: 3fff1b40 40107370 3fff5b84 00000008
[DEBUG] 3fff1a90: 00000000 4000444e 3fff1ad5 40215e24
[DEBUG] 3fff1aa0: 00000000 4000422e 60000200 3fff6a28
[DEBUG] 3fff1ab0: 00000100 40004b14 00000400 003fd000
[DEBUG] 3fff1ac0: 00000100 3fff65a4 3fffc718 003fd400
[DEBUG] 3fff1ad0: 00000002 4000410f 00000833 3fff1bd0
[DEBUG] 3fff1ae0: 3fffc718 40004a3c 000003ff 402686f0
[DEBUG] 3fff1af0: 3fffc718 40104dce 000003ff 40107714
[DEBUG] 3fff1b00: 40230d96 000003ff 40230d48 40230d31
[DEBUG] 3fff1b10: 3fff535c 40230e59 3fff65a4 000003ff
[DEBUG] 3fff1b20: 000003fd 3fff1bd0 3fff1b30 000003fd
[DEBUG] 3fff1b30: ffffff00 55aa55aa 00000033 0000001c
[DEBUG] 3fff1b40: 0000001c 000000cd 00000078 000003ff
[DEBUG] 3fff1b50: 40231208 3fff65a4 3fff65a4 00000001
[DEBUG] 3fff1b60: 00000001 3fff1bf0 4023136d 3ffef5c0
[DEBUG] 3fff1b70: 3fff65a4 00000001 3fff1bd0 00000000
[DEBUG] 3fff1b80: 3fff6664 3fff1c31 00000001 40231420
[DEBUG] 3fff1b90: 3fff1bd0 3fff65a4 3ffe8410 00000001
[DEBUG] 3fff1ba0: 3fff1bf0 3fff65a4 3fff65a4 3ffe8410
[DEBUG] 3fff1bb0: 4023145c 3fff23d0 00000000 3fff0c98
[DEBUG] 3fff1bc0: 40204ef4 3fff0904 00000001 402180fc
[DEBUG] 3fff1bd0: 6401a800 4021fffc 3fff1c50 40218b46
[DEBUG] 3fff1be0: 3fff1db4 3fff0b98 0000000c 40225fc6
[DEBUG] 3fff1bf0: 00000000 00000000 3fff1c40 4010068c
[DEBUG] 3fff1c00: 3fff297c 402050c4 3fff1c50 40202d28
[DEBUG] 3fff1c10: 3fffdad0 3fff0774 00000000 4020c0cf
[DEBUG] 3fff1c20: 401047b2 0001780b 00000000 40226a84
[DEBUG] 3fff1c30: 40230a86 3fff23d0 00000001 3ffe8410
[DEBUG] 3fff1c40: 4020496c 3fff24f8 3fff23d0 40211dfa
[DEBUG] 3fff1c50: 3ffef340

fork04, 2018-02-03:
[788533] [DEBUG] Crash at 33313 ms
[788533] [DEBUG] Reason of restart: 2
[788533] [DEBUG] Exception cause: 0
[788534] [DEBUG] epc1=0x40202f50 epc2=0x00000000 epc3=0x00000000
[788537] [DEBUG] excvaddr=0x00000000 depc=0x00000000
[788541] [DEBUG] >>>stack>>>
[DEBUG] 3ffffd80: 40107123 00000030 0000001e ffffffe0
[DEBUG] 3ffffd90: b7c647c1 00000020 00000004 0000001a
[DEBUG] 3ffffda0: 0000001e 00000004 00000004 00000001
[DEBUG] 3ffffdb0: 20037025 00000002 00000000 40107635
[DEBUG] 3ffffdc0: 401070d0 20037025 00000001 4010711e
[DEBUG] 3ffffdd0: ffffffff 00000020 00000002 4023ae84
[DEBUG] 3ffffde0: ffffffba 00000000 40240000 00000022
[DEBUG] 3ffffdf0: 3fffc200 40107098 3fffc258 4000050c
[DEBUG] 3ffffe00: 4000449a 00000030 00000016 ffffffff
[DEBUG] 3ffffe10: 3feffe00 3fffc718 04000002 003fc2f4
[DEBUG] 3ffffe20: 000002f4 fffffeff 3fff6ef4 00e8a101
[DEBUG] 3ffffe30: fffdffff 3fffc6fc 3fff1d10 3fff6ef4
[DEBUG] 3ffffe40: 000002f4 003fc000 60000600 00000030
[DEBUG] 3ffffe50: 5f6c6f48 47342e32 40007a48 00000001
[DEBUG] 3ffffe60: 20037025 00000002 00000000 40107635
[DEBUG] 3ffffe70: 40107134 3fff0320 00000000 4010711e
[DEBUG] 3ffffe80: ffffffff 00000020 3fff2994 00000000
[DEBUG] 3ffffe90: 00000000 00000000 0000001f 401048fd
[DEBUG] 3ffffea0: 4000050c 40107098 3fffc258 4000050c
[DEBUG] 3ffffeb0: 40000f68 00000030 00000014 ffffffff
[DEBUG] 3ffffec0: 40000f58 00000000 00000020 00000000
[DEBUG] 3ffffed0: feefeffe 3fff1d10 3fff1d10 fffffffe
[DEBUG] 3ffffee0: ffffffff 3ffef31c 3fff1d10 3fffdab0
[DEBUG] 3ffffef0: 00000000 3fffdad0 3fff0c98 00000030
[DEBUG] 3fffff00: 00000000 400042db 3ffed140 60000600
[DEBUG] 3fffff10: 40004b31 3fff6ef4 000002f4 003fc000
[DEBUG] 3fffff20: 40104e62 3ffef340 3ffef1a0 40107714
[DEBUG] 3fffff30: 4022c641 3ffef1a0 3ffef340 01f8a317
[DEBUG] 3fffff40: 3fff6ef4 00001000 4022cade 00000008
[DEBUG] 3fffff50: 00000000 00000000 4022cb8b 3ffef254
[DEBUG] 3fffff60: 3ffef340 009d8c98 4022caa0 402036c4
[DEBUG] 3fffff70: 4022f85d 3ffef254 3ffef340 01f886fe
[DEBUG] 3fffff80: 4022f8a2 3fffdab0 00000000 3fffdcb0
[DEBUG] 3fffff90: 3ffef358 3fffdad0 3fff0c98 40202e7f
[DEBUG] 3fffffa0: 40000f49 000314d8 3fffdab0 40000f49
[DEBUG] <<<stack<<<
[788713] +OK

xoseperez, 2018-02-03:
This is a known issue due to the ESPSoftwareSerial library. There are a lot of reports in the library repository about it (https://github.com/plerup/espsoftwareserial/issues). I'm trying to find a fix.

xoseperez, 2018-03-19:
:smile: