xoseperez/espurna #516: 5-channel dimmer on H801 is discovered by HomeAssistant as RGB + Switch

Sanfam, 2018-02-03:
I have several H801 modules being successfully discovered by homeassistant.  While they offer 5 individual dimmer channels, they instead register as a combination of one dimmable composite RGBW device and one switchable load.  

Second, and this may be an issue with HomeAssistant itself, but the discovered device links the RGB setting to global power and the W1 channel as a secondary.  If RGB is dimmed to 0, the global power state is switched off regardless of what the W1 channel is set to, turning off all channels and hiding both controls.

Overview:
[image]

RGB+W1 Device:
[image]

5th Channel (W2):
[image]

xoseperez, 2018-03-02:
The switchable load is actually created by HA itself when it discovers a wemo-compatible device. You probably have Alexa integration enabled in your ESPurna device and, Apparently, the `wemo` component in home assistant is enabled by default. Check the HA log for something like this:

```
INFO:homeassistant.components.discovery:Found new service: belkin_wemo {'ssdp_description': 'http://192.168.1.10:52000/setup.xml', 'model_number': '1.0', 'mac_address': '', 'model_name': 'Socket', 'host': '192.168.1.10', 'serial': '221703K0B93C93', 'port': 52000, 'name': 'MINI'}
INFO:homeassistant.loader:Loaded wemo from homeassistant.components.wemo
INFO:homeassistant.setup:Setting up wemo
INFO:pywemo.subscribe:Listening on port 8989
INFO:homeassistant.components.wemo:Scanning for WeMo devices.
WARNING:homeassistant.components.wemo:Unable to probe wemo at 192.168.1.10
```

And you are right. Setting the brightness to 0 turns OFF everything. As far as I can tell, this is a "feature" in Home Assistant. Whenever the brightness value is set to 0 it also send an OFF payload to the state_command, and when it's non-zero it sends an ON...

joeltetava, 2018-04-25:
Is there a website that has the instructions to this?

Zixim, 2018-04-25:
@joeltetava A link to instructions for configuring the H801 in Home Assistant ?
https://github.com/xoseperez/espurna/wiki/HomeAssistant#link-an-espurna-light-to-home-assistant

closed 2021-10-03 by mcspr