xoseperez/espurna #519: Sonoff Basic Won't Retain Configuration

DavidPower, 2018-02-03:
@xoseperez,

I've flashed a Sonoff Basic device with v 1.12.3.

I can access the web interface, update the password, configure Wi-Fi, save changes, etc.

But as soon as I reboot the device via either the web interface or by power-cycling, it loses all configuration details and I have to start from scratch again... even the admin password is lost.

The device never seems to leave setup mode (i.e. green led flashing).

Is there something I'm missing?

Thanks,
dp

xoseperez, 2018-02-03:
Did you flash it with the right flash size? Can you copy here the debug log of the device booting?

DavidPower, 2018-02-03:
Based on your question, I went back through the docs and determined I may have goofed with the flash size. I read the basic as having an 8M byte memory and flashed as "-fs 8MB." 

In fact, it's 8M "bit"... so I should have flashed with "-fs 1MB."

At any rate, I'm happy to copy the debug log. How do I access that?

DavidPower, 2018-02-03:
Sure enough... I went back and reflashed the device with the proper size (i.e. 1MB) and now everything's working as it should.

Thank you for the lead @xoseperez.

You can close this issue at your convenience.

xoseperez, 2018-02-04:
Good! Enjoy!

closed 2018-02-04 by xoseperez