xoseperez/espurna #521: Fast MQTT commands are ignored

hyteoo, 2018-02-04:
I want to reopen this issue here as I see there is really an issue with MQTT
After playing around with the RELAY_FLOOD_WINDOW and RELAY_FLOOD_CHANGES, the behavior is the same.

MQTT commands sent less than 2000ms apart are ignored.
This was tested on several builds and devices.

Double checked if there's a mqtt /network issue however that is not the case.

example: running this
mosquitto_pub -t testdevice/relay/0/set -m 2
timeout 1
mosquitto_pub -t testdevice/relay/0/set -m 2
timeout 1
(...repeat about 20 times)

the relay switches only once every 2-3 seconds.

Running this

mosquitto_pub -t testdevice/relay/0/set -m 2
mosquitto_pub -t testdevice/relay/0/set -m 2
mosquitto_pub -t testdevice/relay/0/set -m 2
mosquitto_pub -t testdevice/relay/0/set -m 2
(.... repeat about 100 times)

the relay only switches once every 2 seconds.

xoseperez, 2018-02-04:
There could be a number of reasons for MQTT no responding fast enough, including network latency and application polling. But I'd like to know the purpose of this test...

hyteoo, 2018-02-05:
There's no specific purpose except the one that the ESP is not responsive (or at least doesn't seem to be responsive)
Network latency shouldn't be an issue, as the first command is executed in a matter of 10's of ms.

Is there any other parameter I should look after?

hyteoo, 2018-02-06:
Update

after further investigation, I see that somehow consecutive commands are received by espurna but it's weird how they are treated.

After sending 10 commands over MQTT:
-t Relay1/relay/0/set -m 2

I get this:

[412946] [MQTT] Received Relay1/relay/0/set => 2
[412948] [RELAY] #0 scheduled ON in 0 ms
[412951] [RELAY] #0 set to ON
[412952] [MQTT] Sending Relay1/relay/0 => 1 (PID 88)
[412964] [MQTT] Received Relay1/relay/0/set => 2
[412965] [RELAY] #0 scheduled OFF in 0 ms
[412966] [RELAY] #0 set to OFF
[412967] [MQTT] Sending Relay1/relay/0 => 0 (PID 89)
[412978] [MQTT] Publish ACK for PID 88
[413968] [RELAY] Saving mask: 0
[415399] [MQTT] Publish ACK for PID 89
[415400] [MQTT] Received Relay1/relay/0/set => 2
[415401] [RELAY] #0 scheduled ON in 0 ms
[415402] [MQTT] Received Relay1/relay/0/set => 2
[415402] [RELAY] #0 scheduled ON in 0 ms
[415403] [MQTT] Received Relay1/relay/0/set => 2
[415403] [RELAY] #0 scheduled ON in 0 ms
[415403] [MQTT] Received Relay1/relay/0/set => 2
[415404] [RELAY] #0 scheduled ON in 0 ms
[415404] [MQTT] Received Relay1/relay/0/set => 2
[415405] [RELAY] #0 scheduled ON in 0 ms
[415405] [MQTT] Received Relay1/relay/0/set => 2
[415406] [RELAY] #0 scheduled ON in 0 ms
[415406] [MQTT] Received Relay1/relay/0/set => 2
[415407] [RELAY] #0 scheduled ON in 0 ms
[415407] [MQTT] Received Relay1/relay/0/set => 2
[415408] [RELAY] #0 scheduled ON in 0 ms
[415409] [RELAY] #0 set to ON
[415410] [MQTT] Sending Relay1/relay/0 => 1 (PID 90)
[415424] [MQTT] Publish ACK for PID 90
[416411] [RELAY] Saving mask: 1

Even though 2 is received, the relay is repeatedly set to ON, resulting in no action.

xoseperez, 2018-02-06:
ESPurna uses an async MQTT library (https://github.com/marvinroger/async-mqtt-client/) and changes to the relays are actually enqueue by the flood window feature. Relay structure hold the current and the target value for the relay. It could be that a new message is processed before the relay has been toggled. And 2 always means change "current" state. If the value of current has not been changed yet target would remain the same.

hyteoo, 2018-02-06:
That's a good point
So I tried this:

mosquitto_pub -h 192.168.0.60 -t Relay1/relay/0/set -m 0
mosquitto_pub -h 192.168.0.60 -t Relay1/relay/0/set -m 1
mosquitto_pub -h 192.168.0.60 -t Relay1/relay/0/set -m 0
mosquitto_pub -h 192.168.0.60 -t Relay1/relay/0/set -m 1
mosquitto_pub -h 192.168.0.60 -t Relay1/relay/0/set -m 0
mosquitto_pub -h 192.168.0.60 -t Relay1/relay/0/set -m 1
mosquitto_pub -h 192.168.0.60 -t Relay1/relay/0/set -m 0
mosquitto_pub -h 192.168.0.60 -t Relay1/relay/0/set -m 1
mosquitto_pub -h 192.168.0.60 -t Relay1/relay/0/set -m 0

Result is different:

[581617] [MQTT] Received Relay1/relay/0/set => 1
[581618] [RELAY] #0 scheduled ON in 0 ms
[581619] [MQTT] Received Relay1/relay/0/set => 0
[581619] [RELAY] #0 scheduled change cancelled
[581620] [MQTT] Received Relay1/relay/0/set => 1
[581620] [RELAY] #0 scheduled ON in 0 ms
[581621] [MQTT] Received Relay1/relay/0/set => 0
[581621] [RELAY] #0 scheduled change cancelled
[581621] [MQTT] Received Relay1/relay/0/set => 1
[581622] [RELAY] #0 scheduled ON in 0 ms
[581622] [MQTT] Received Relay1/relay/0/set => 0
[581623] [RELAY] #0 scheduled change cancelled
[581623] [MQTT] Received Relay1/relay/0/set => 1
[581623] [RELAY] #0 scheduled ON in 0 ms
[581624] [MQTT] Received Relay1/relay/0/set => 0
[581624] [RELAY] #0 scheduled change cancelled
[581625] [MQTT] Received Relay1/relay/0/set => 1
[581625] [RELAY] #0 scheduled ON in 0 ms
[581626] [MQTT] Received Relay1/relay/0/set => 0
[581626] [RELAY] #0 scheduled change cancelled
[581626] [MQTT] Received Relay1/relay/0/set => 1
[581627] [RELAY] #0 scheduled ON in 0 ms
[581627] [MQTT] Received Relay1/relay/0/set => 0
[581628] [RELAY] #0 scheduled change cancelled
[581628] [MQTT] Received Relay1/relay/0/set => 1
[581628] [RELAY] #0 scheduled ON in 0 ms
[581629] [MQTT] Received Relay1/relay/0/set => 0
[581629] [RELAY] #0 scheduled change cancelled
[581630] [MQTT] Received Relay1/relay/0/set => 1
[581630] [RELAY] #0 scheduled ON in 0 ms
[581631] [RELAY] #0 set to ON
[581632] [MQTT] Sending Relay1/relay/0 => 1 (PID 279)
[581644] [MQTT] Publish ACK for PID 279
[582633] [RELAY] Saving mask: 1

Did you have a chance to test this? As mentioned before, it actually works after reboot. Only after a few minutes of inactivity it starts to lag.

hyteoo, 2018-02-06:
I ran a sequence with 1 second delay in between:
mosquitto_pub -h 192.168.0.60 -t Relay1/relay/0/set -m 0
timeout 1
mosquitto_pub -h 192.168.0.60 -t Relay1/relay/0/set -m 1
timeout 1
mosquitto_pub -h 192.168.0.60 -t Relay1/relay/0/set -m 0
timeout 1
mosquitto_pub -h 192.168.0.60 -t Relay1/relay/0/set -m 1
timeout 1
mosquitto_pub -h 192.168.0.60 -t Relay1/relay/0/set -m 0
timeout 1
mosquitto_pub -h 192.168.0.60 -t Relay1/relay/0/set -m 1
timeout 1
mosquitto_pub -h 192.168.0.60 -t Relay1/relay/0/set -m 0
timeout 1
mosquitto_pub -h 192.168.0.60 -t Relay1/relay/0/set -m 1
timeout 1
mosquitto_pub -h 192.168.0.60 -t Relay1/relay/0/set -m 0
timeout 1
mosquitto_pub -h 192.168.0.60 -t Relay1/relay/0/set -m 1
timeout 1

result more or less the same:
[818943] [MQTT] Received Relay1/relay/0/set => 0
[818945] [RELAY] #0 scheduled OFF in 0 ms
[818946] [RELAY] #0 set to OFF
[818947] [MQTT] Sending Relay1/relay/0 => 0 (PID 284)
[818954] [MQTT] Publish ACK for PID 284
[819948] [RELAY] Saving mask: 0
[821617] [MQTT] Received Relay1/relay/0/set => 1
[821619] [RELAY] #0 scheduled ON in 0 ms
[821620] [MQTT] Received Relay1/relay/0/set => 0
[821620] [RELAY] #0 scheduled change cancelled
[824617] [MQTT] Received Relay1/relay/0/set => 1
[824619] [RELAY] #0 scheduled ON in 0 ms
[824620] [MQTT] Received Relay1/relay/0/set => 0
[824620] [RELAY] #0 scheduled change cancelled
[824621] [MQTT] Received Relay1/relay/0/set => 1
[824621] [RELAY] #0 scheduled ON in 0 ms
[824624] [RELAY] #0 set to ON
[824625] [MQTT] Sending Relay1/relay/0 => 1 (PID 285)
[824632] [MQTT] Publish ACK for PID 285
[825626] [RELAY] Saving mask: 1
[827618] [MQTT] Received Relay1/relay/0/set => 0
[827619] [RELAY] #0 scheduled OFF in 0 ms
[827620] [MQTT] Received Relay1/relay/0/set => 1
[827620] [RELAY] #0 scheduled change cancelled
[827621] [MQTT] Received Relay1/relay/0/set => 0
[827621] [RELAY] #0 scheduled OFF in 0 ms
[827622] [RELAY] #0 set to OFF
[827623] [MQTT] Sending Relay1/relay/0 => 0 (PID 286)
[827632] [MQTT] Publish ACK for PID 286
[828624] [RELAY] Saving mask: 0
[830618] [MQTT] Received Relay1/relay/0/set => 1
[830620] [RELAY] #0 scheduled ON in 0 ms
[830622] [RELAY] #0 set to ON
[830623] [MQTT] Sending Relay1/relay/0 => 1 (PID 287)
[830631] [MQTT] Publish ACK for PID 287
[831624] [RELAY] Saving mask: 1

xoseperez, 2018-02-07:
These logs look perfectly normal. If the new message is received before the relay is actually toggles then the ac tion is cancelled, so this pattern is normal:

```
[581617] [MQTT] Received Relay1/relay/0/set => 1
[581618] [RELAY] #0 scheduled ON in 0 ms
[581619] [MQTT] Received Relay1/relay/0/set => 0
[581619] [RELAY] #0 scheduled change cancelled
```

If you leave enough time the relay is correctly toggled:

```
[824621] [MQTT] Received Relay1/relay/0/set => 1
[824621] [RELAY] #0 scheduled ON in 0 ms
[824624] [RELAY] #0 set to ON
[824625] [MQTT] Sending Relay1/relay/0 => 1 (PID 285)
[824632] [MQTT] Publish ACK for PID 285
[825626] [RELAY] Saving mask: 1
```

Mind this is not meant to do fast-switching. You probably need to hack the relay module and remove flood window and maybe use PubSubClient instead of AsycnMQTTClient if you really need fast refresh.

hyteoo, 2018-02-07:
Ok, I will look some more into this. On the MQTT side things seem to work correctly, messages are received.
Also switching from the web GUI or external HW switch works as intended (fast)
So there has to be something between MQTT processing and the relay command.

I still think there's a bug (or a feature?) somewhere because after boot things run fast. After 2-3 minutes, laggy. So there must be a timer or something in the middle.
Any suggestions would be appreciated :)

Thanks!

xoseperez, 2018-02-08:
Do you have anything else enabled? Usually, it's the other way around: booting is when the devices do most heavy lifting things...

hyteoo, 2018-02-08:
Most of the services are disabled, I only use MQTT in my network.
Here's a backup from one of the S20 switches - flashed with the release .bin from github:

`{"app":"ESPURNA","version":"1.12.3","relayPulse0":"0","cfg":"3","board":"13","ledGPIO0":"13","ledLogic0":"1","btnGPIO0":"0","btnRelay0":"0","relayGPIO0":"12","relayType0":"0","adminPass":"---","haPrefix":"homeassistant","relaySync":"0","relayTime0":"0","mqttGroupInv0":"0","useColor":"0","useRGB":"0","useWhite":"0","useGamma":"0","useCSS":"0","useTransitions":"0","apiEnabled":"0","apiRealTime":"0","apiKey":"A62A55515E630474","telnetSTA":"0","nofussEnabled":"0","ssid0":"---","pass0":"---","mask0":"255.255.255.0","dns0":"8.8.8.8","mqttPort":"1883","mqttKeep":"30","mqttUseSSL":"0","mqttTopic":"{identifier}","ntpDST":"1","dczEnabled":"0","dczSkip":"2","dczTopicIn":"domoticz/in","dczTopicOut":"domoticz/out","dczRelayIdx0":"0","idbEnabled":"0","snsRead":"6","tmpUnits":"0","tmpCorrection":"0","pwrResetCalibration":"0","mqttEnabled":"1","mqttServer":"192.168.0.60","hostname":"SwitchSocket1","ledMode0":"1","relayBoot0":"2","alexaEnabled":"0","tspkEnabled":"0","tspkRelay0":"0","mqttQoS":"2","mqttRetain":"1","wifiScan":"0","btnDelay":"200","haEnabled":"0","boardName":"ITEAD_S20","ntpServer":"pool.ntp.org","ntpOffset":"120","pwrExpectedC":"","pwrExpectedV":"","pwrExpectedP":"","mqttUseJson":"0","now":"2018-02-01 22:04:43"}
`

xoseperez, 2018-02-09:
Sorry, no useful info there... is it only MQTT that is "affected" by this "slowness"? The web UI works fine? Telnet?

hyteoo, 2018-02-09:
Only MQTT, yes.
Web UI - instant
Telnet also works fine, no visible lag.
As saw in my previous posts, the messages are received by the MQTT processing but the command is blocked somewhere on the line between the MQTT interpreter and relay cmd.
I also looked through the code however haven't found any plausible reason.
It's like the relay debounces the commands from MQTT by 2 sec.

And again... testing for ex. 30sec after reboot, MQTT is instant. Leave it 2-3 more minutes, becomes laggy

hyteoo, 2018-03-01:
Anybody else has this issue?

esalii, 2018-03-12:
I have the same issue and it exists since fw 1.9.9 but it has not been fixed yet.
That is the reason i still use the old fw because for me it is very important that mqtt works fast.

closed 2018-08-09 by stale[bot]