xoseperez/espurna #522: Newbie : Can not compile

franki29, 2018-02-04:
Hello,
i try to compile the itead-sonoff-rfbridge but get error : 
Arduino\libraries\ESPAsyncWebServer-master\src/ESPAsyncWebServer.h:62:18: error: redeclaration of 'HTTP_ANY'

   HTTP_ANY     = 0b01111111,

No matter if i use Platform IO or Arduino IDE.
Seems that problem was existing also in the past, but i can not figure out what to do.

The prebuild image does not work because i upload the https://github.com/rhx/RF-Bridge-EFM8BB1 custom firmware. I can receive code from my remote controll, but it does not send a code out, so i thougt i have to change the settings for the sending commands.

Thanks for helping
Frank

xoseperez, 2018-02-04:
Does it say when it was first declared? Are you building the firmware without modifications?

franki29, 2018-02-04:
I did not change the settings. And it does not matter what firmware i like to build.

Here is the full error message. 

rduino: 1.8.5 (Windows 10), Board: "Generic ESP8266 Module, 80 MHz, 40MHz, DIO, 115200, 512K (64K SPIFFS), ck, Disabled, None"

In file included from sketch\config/prototypes.h:13:0,

                 from sketch\config/all.h:20,

                 from C:\Users\frank\Downloads\espurna-1.12.3\espurna-1.12.3\code\espurna\espurna.ino:24:

C:\Users\frank\Documents\Arduino\libraries\ESPAsyncWebServer-master\src/ESPAsyncWebServer.h:55:18: error: redeclaration of 'HTTP_GET'

   HTTP_GET     = 0b00000001,

                  ^

In file included from C:\Users\frank\Downloads\espurna-1.12.3\espurna-1.12.3\code\espurna\espurna.ino:1:0:

C:\Users\frank\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WebServer\src/ESP8266WebServer.h:30:29: note: previous declaration 'HTTPMethod HTTP_GET'

 enum HTTPMethod { HTTP_ANY, HTTP_GET, HTTP_POST, HTTP_PUT, HTTP_PATCH, HTTP_DELETE, HTTP_OPTIONS };

                             ^

In file included from sketch\config/prototypes.h:13:0,

                 from sketch\config/all.h:20,

                 from C:\Users\frank\Downloads\espurna-1.12.3\espurna-1.12.3\code\espurna\espurna.ino:24:

C:\Users\frank\Documents\Arduino\libraries\ESPAsyncWebServer-master\src/ESPAsyncWebServer.h:56:18: error: redeclaration of 'HTTP_POST'

   HTTP_POST    = 0b00000010,

                  ^

In file included from C:\Users\frank\Downloads\espurna-1.12.3\espurna-1.12.3\code\espurna\espurna.ino:1:0:

C:\Users\frank\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WebServer\src/ESP8266WebServer.h:30:39: note: previous declaration 'HTTPMethod HTTP_POST'

 enum HTTPMethod { HTTP_ANY, HTTP_GET, HTTP_POST, HTTP_PUT, HTTP_PATCH, HTTP_DELETE, HTTP_OPTIONS };

                                       ^

In file included from sketch\config/prototypes.h:13:0,

                 from sketch\config/all.h:20,

                 from C:\Users\frank\Downloads\espurna-1.12.3\espurna-1.12.3\code\espurna\espurna.ino:24:

C:\Users\frank\Documents\Arduino\libraries\ESPAsyncWebServer-master\src/ESPAsyncWebServer.h:57:18: error: redeclaration of 'HTTP_DELETE'

   HTTP_DELETE  = 0b00000100,

                  ^

In file included from C:\Users\frank\Downloads\espurna-1.12.3\espurna-1.12.3\code\espurna\espurna.ino:1:0:

C:\Users\frank\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WebServer\src/ESP8266WebServer.h:30:72: note: previous declaration 'HTTPMethod HTTP_DELETE'

 enum HTTPMethod { HTTP_ANY, HTTP_GET, HTTP_POST, HTTP_PUT, HTTP_PATCH, HTTP_DELETE, HTTP_OPTIONS };

                                                                        ^

In file included from sketch\config/prototypes.h:13:0,

                 from sketch\config/all.h:20,

[... 53 lines cut ...]

                                                                                     ^

In file included from sketch\config/prototypes.h:13:0,

                 from sketch\config/all.h:20,

                 from C:\Users\frank\Downloads\espurna-1.12.3\espurna-1.12.3\code\espurna\espurna.ino:24:

C:\Users\frank\Documents\Arduino\libraries\ESPAsyncWebServer-master\src/ESPAsyncWebServer.h:62:18: error: redeclaration of 'HTTP_ANY'

   HTTP_ANY     = 0b01111111,

                  ^

In file included from C:\Users\frank\Downloads\espurna-1.12.3\espurna-1.12.3\code\espurna\espurna.ino:1:0:

C:\Users\frank\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WebServer\src/ESP8266WebServer.h:30:19: note: previous declaration 'HTTPMethod HTTP_ANY'

 enum HTTPMethod { HTTP_ANY, HTTP_GET, HTTP_POST, HTTP_PUT, HTTP_PATCH, HTTP_DELETE, HTTP_OPTIONS };

                   ^

exit status 1
Fehler beim Kompilieren für das Board Generic ESP8266 Module.

franki29, 2018-02-08:
Sorry, by mistake i closed the issue, any help what i doing wong to compile?

xoseperez, 2018-02-09:
It says you have two declarations of the same constants (HTTP_*) in files

`C:\Users\frank\Documents\Arduino\libraries\ESPAsyncWebServer-master\src/ESPAsyncWebServer.h`
`C:\Users\frank\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WebServer\src/ESP8266WebServer.h`

Apparently, this is due to the Arduino IDE not filtering the libraries the project uses (the seconds one should no be built). You are using Arduino IDE 1.5? Try upgrading to 1.8.X.

franki29, 2018-02-09:
Hi,
i am using 1.8.5 the newest one i could get in the internet. I download everything new and made a fresh installation. I found that in my old espurna.ino a entry in the first line (You can see it in my error message), that is not in the new one from today and the compiling is working.
Thanks and best regards

closed 2018-02-09 by franki29