xoseperez/espurna #523: Reading kWh from POW

Dagobert1966, 2018-02-04:
I have a POW to register the used power to charge my car.
The POW is comunicating with Domoticz, but I cannot seem to read the proper kW and kWh in Domoticz.
I've setup a virtual meter : Electric, Power and counter.
It seems to register the Energy #0 sensor of the POW.
How can I setup to measure kW and kWh?

fork04, 2018-02-05:
Energy #0 is energy USED from start of Sonoff-Pow in  Joules - https://en.wikipedia.org/wiki/Joule
[image]
You can map IDX here ^^^

xoseperez, 2018-02-05:
Active power gives you the current power consumption in W and, as @fork04  says, the energy field gives you the total aggregated energy since the device started in Joules (1kWh = 3600000J).

Dagobert1966, 2018-02-05:
Thanks all for the replies.
OK, so energy #0 is in Joules.
But how can I convert this in Domoticz in kWh?

xoseperez, 2018-02-05:
Sorry, I can't help with that, I don't use Domoticz myself. The reason to use Joules is that that's the unit of energy in the IS.

Dagobert1966, 2018-02-05:
Thanks xoseperez.
Maybe someone else can help?

digitalesPL, 2018-02-22:
Id like this also for Home Assistant i guess its a simple maths inside the software? And some data output? I think there is suport for this in Tasmota?

digitalesPL, 2018-02-22:
How to get to this information? I saw this on some other page regarding Sonoff POW and Tasmota software.
The presented information has the following meaning:
Message   | Unit | Description
----------|------|-----------------------------------------------------
Total     | kWh  | Total Energy usage including Today
Yesterday | kWh  | Total Energy usage between 00:00 and 24:00 yesterday
Today     | kWh  | Total Energy usage today from 00:00 until now
Period    | Wh   | Energy usage between previous message and now
Power     | W    | Current power load
Factor    |      | The ratio of the real power flowing to the load to
          |      |   the apparent power in the circuit 
Voltage   | V    | Current line voltage
Current   | A    | Current line current

https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-Pow

xoseperez, 2018-02-26:
I have added the option to select between joules and kWh for energy reporting in the web interface. It's in the dev branch.

xoseperez, 2018-02-26:
This is staged for release.

closed 2018-02-26 by xoseperez

foncubierta, 2018-04-22:
Can you get the monthly power consumed?

xoseperez, 2018-04-23:
No. If you need this kind of info I recommend you to log the minute consumptions to a database like InfluxDB and the use Grafana to graph daily/monthly/... consumptions