xoseperez/espurna #524: Wemos D1 Mini Builds with Sensors included

rradar, 2018-02-05:
Hello!

I'm new with espurna and flashed successful (pre build images) to two devices (itead s20 & wemos d1 mini). The s20 is working as intended but I don't get my wemos d1 mini with temperature/humidity sensor running. I read the new wiki article for sensors here: https://github.com/xoseperez/espurna/wiki/Sensors

As I understand espurna needs to be compiled/builded to be used with sensors. The problem is that I don't have these skills. I wonder if there is a possible to provide prebuild "Wemos D1 Mini" images with support for sensors included. The available images are less than 500kb in size and all wemos d1 mini's have a 4MB of flash. So there's plenty of space!

Thank's for the great work of espurna. It's relay nice to use!

xoseperez, 2018-02-05:
Is not that simple, right now the sensor configuration (GPIO, types,...) is also defined as build settings. So first I should add a section to configure them in the web UI. It's on the roadmap, hopefully soon.

xoseperez, 2018-07-01:
See #382 and project https://github.com/xoseperez/espurna/projects/2

closed 2018-07-01 by xoseperez