xoseperez/espurna #525: issue with push button behaviour sonoff dual

jmphihpi, 2018-02-05:
the control off the relays doesn't respect the duration value setup in the GUI when controled with a push button  between pins ground, button0 and button1

xoseperez, 2018-02-05:
Sorry, do you mean the pulse duration does not work when using button0 or button1 in the Dual header?

jmphihpi, 2018-02-05:
Thanks for your interest
Yes, absolutely, despite double click delay =100ms and pulse duration is 20 sec

xoseperez, 2018-02-05:
The double-click won't work on a Dual (v1) since there is a secondary microcontroller driving the buttons. But the pulse duration should work. What does it do, exactly?

jmphihpi, 2018-02-05:
for instance, if you connect a push button between gnd and button0, the relay will close as long as the button is pressed and not for the duration of the" pulse duration" parameter

xoseperez, 2018-02-06:
Do you have a Sonoff Dual R2? ESPurna defines button0 and button1 on the Dual R2 as switches, so they toggle on both edges. You might want to change the definition of these buttons in the hardware.h file:

https://github.com/xoseperez/espurna/blob/f8907d60395ccfe0d606b4fb24624720fbcd7f51/code/espurna/config/hardware.h#L389-L390

To be:

```
    #define BUTTON1_MODE        BUTTON_PUSHBUTTON | BUTTON_SET_PULLUP | BUTTON_DEFAULT_HIGH
    #define BUTTON2_MODE        BUTTON_PUSHBUTTON | BUTTON_SET_PULLUP | BUTTON_DEFAULT_HIGH
```

jmphihpi, 2018-02-06:
Great! thanks a lot. I will test asap
JM

closed 2018-02-06 by jmphihpi

ouadie2010, 2019-03-11:
Hello, 
I'm getting the same issue with the buttons, they are not respecting the pulse duration. Could somebody share the bin file with the "press button" function