xoseperez/espurna #526: Unable to flash sonoff basic (Timed out waiting for packet header)

lyghtnox, 2018-02-05:
Hello,
I tried to flash my sonoff basic but I get the message : Timed out waiting for packet header.
I tried to flash it with a CH340 chipset.
Here is the entire error message:

`Connecting........_____....._____....._____....._____....`
`Detecting chip type... ESP8266`
`Chip is ESP8266EX`
`Uploading stub...`
`Running stub...`
`Stub running...`
`Configuring flash size...`
`Auto-detected Flash size: 1MB`
`Compressed 495808 bytes to 341889...`
`Writing at 0x00008000... (14 %)`
`A fatal error occurred: Timed out waiting for packet header`

How can I solve it?

xoseperez, 2018-02-05:
Does it always happen? Sometimes retrying is just enough.

lyghtnox, 2018-02-05:
Yeah it happens every time. Always stopping at 14%

xoseperez, 2018-02-05:
I can't tell. It could be a hardware problem... have you flashed it before?

lyghtnox, 2018-02-05:
No it's the first time I try. That's sad

bopnrgi, 2018-02-06:
Try to change baud rate transfert, for example with this rate : 38400
It solves the problem for me (flashing takes several minutes in this case but works).
It appears only on one device for me, others works at "high speed transfert"

lyghtnox, 2018-02-07:
Thanks for the answer bopnrgi but that doesn't work either. I'll open an issue on the esptool github.

lyghtnox, 2018-02-08:
I tried with a lower baudrate (38400) and I got the same error but this time at 0x00018000

lyghtnox, 2018-02-10:
This was a software problem. The sonoff-basic firmware gave me the error but the generic and the basic-dallas firmware worked fine. Thank for the answers!

closed 2018-02-10 by lyghtnox