xoseperez/espurna #528: Can't differentiate the temperature values of 2 sensors in InfluxDB

Rapejim, 2018-02-06:
I have connected two sensors (Dallas -> GPIO12, and DHT11 -> GPIO12) on a NodeMCU with ESPurna 1.12.3.
And I have configured InfluxDB support well.

But the system save in database, both **_temperature_** values under the same "device"...
[image]

and it is impossible differentiate each sensor:
[image]
For example, using Grafana, it do the average of both values always (except when I have disconnected one of sensors).

xoseperez, 2018-02-06:
There should be another attribute `id` that differentiates both magnitudes.

closed 2018-04-10 by Rapejim

djsomi, 2020-06-11:
Finally we found the solution: device and ID should be definied at the end of the grafana query:

[image]