xoseperez/espurna #529: UART --> MQTT

beho1der, 2018-02-06:
Is it possible to add new functionality(UART --> MQTT): from the terminal, the publicat of the data in the mqtt topic. It is necessary for interaction with other devices
Example: 
send data to serial --> #mqttmydevice test-string
mqtt <---  mqtt_root_topic/mydevice/test-string

xoseperez, 2018-02-06:
I like this one. It would be a great addition to use with RFID or barcode readers...

AlbertWeterings, 2018-02-07:
Hi,

I created something like this data going from UART to MQTT. As I only needed a quick and dirty working approach so I high jacked the RF-Bridge code for it. I will attache the file so you have an idea.
[kwhalbert.zip](https://github.com/xoseperez/espurna/files/1702177/kwhalbert.zip)

AlbertWeterings, 2018-02-19:
I tried to make a Pull request sorry that went wrong.
 I was working on this enhancement it works partially and need some help. With these code changes you will be able to receive data via UART and it will be send to MQTT. The opposite way I can't get to work as somehow I can't get to the data received via MQTT maybe someone knows a solution.

In espurna add on line 150
`    #ifdef UARTtoMQTT
        UARTtoMQTTSetup();
    #endif
`
in Arduino.h add on line 63
`#define UARTtoMQTT`

in general.h add on line 557
`#define MQTT_TOPIC_UARTIN       "uartin"
#define MQTT_TOPIC_UARTOUT      "uartout"
`
in hardware.h add on line 1425
`#elif defined(UARTtoMQTT)

    // Info
    #define MANUFACTURER        "Albert"
    #define DEVICE              "UART_experiment"
    #define SERIAL_BAUDRATE     115200

    // Remove UART noise on serial line
    #define TERMINAL_SUPPORT        0
    #define DEBUG_SERIAL_SUPPORT    0

    // Buttons
    #define BUTTON1_PIN         0
    #define BUTTON1_MODE        BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH

    // LEDs
    #define LED1_PIN            1
    #define LED1_PIN_INVERSE    1`

Add a file uart_mqtt.ino with the following content
`#ifdef UARTtoMQTT

#include <queue>
#include <Ticker.h>

// -----------------------------------------------------------------------------
// GLOBALS TO THE MODULE
// -----------------------------------------------------------------------------
const byte numChars = 100;
char receivedChars[numChars]; // an array to store the received data
boolean newData = false;

void _recvWithEndMarker() {
  static byte ndx = 0;
  char endMarker = '\n';
  char rc;
 
  while (Serial.available() > 0 && newData == false) {
    rc = Serial.read();

    if (rc != endMarker) {
      receivedChars[ndx] = rc;
      ndx++;
      if (ndx >= numChars) {
        ndx = numChars - 1;
      }
    }
    else {
      receivedChars[ndx] = '\0'; // terminate the string
      ndx = 0;
      newData = true;
    }
  }
}

void _sendNewData() {
  if (newData == true && MQTT_SUPPORT) {
    #if MQTT_SUPPORT
       mqttSend(MQTT_TOPIC_UARTIN, receivedChars); // publish: UART -> mqtt bus
    #endif
    newData = false;
  }
}

#if MQTT_SUPPORT
void _UARTtoMQTTMqttCallback(unsigned int type, const char * topic, const char * payload) {
    if (type == MQTT_CONNECT_EVENT) {
        mqttSubscribe(MQTT_TOPIC_UARTOUT);
        DEBUG_MSG_P(PSTR("[UARTtoMQTT] MQTT Subscribe topic:\n"), MQTT_TOPIC_UARTOUT);
    }

    if (type == MQTT_MESSAGE_EVENT) {

        // Match topic
        String t = mqttTopicKey((char *) topic);
        //DEBUG_MSG_P(PSTR("[UARTtoMQTT] t= :\n"), t);

        bool isUARTOut = t.equals(MQTT_TOPIC_UARTOUT);

        if (isUARTOut) {
          //send the received MQTT message to Serial

        }

    }

}
#endif

// -----------------------------------------------------------------------------
// SETUP & LOOP
// -----------------------------------------------------------------------------

void UARTtoMQTTSetup() {

    #if MQTT_SUPPORT
        mqttRegister(_UARTtoMQTTMqttCallback);
    #endif

    // Register oop
    espurnaRegisterLoop(UARTtoMQTTLoop);

}

void UARTtoMQTTLoop() {
     _recvWithEndMarker();
     _sendNewData();
}

#endif
`

It is not the most nice code but it works for one direction. In uart_mqtt.ino you will find a line //send the received MQTT message to Serial from here I got stuck hope someone can help a bit.

beho1der, 2018-02-20:
Thanks, It would be nice to include in the main branch of the project

AlbertWeterings, 2018-02-20:
for now my experiment can be found here:
https://github.com/xoseperez/espurna/compare/master...AlbertWeterings:UARTtoMQTT?expand=1
It still needs work as it only works from UART to MQTT not opposite My programming skills are basic and until now I could not find a way to pass the via MQTT received data to UART.

xoseperez, 2018-02-26:
@AlbertWeterings  I have just merged your code. Thank you! I have made some small changes to follow the code style convention and moved some things up and down. It's in the dev branch now.

xoseperez, 2018-02-26:
Staged for release

closed 2018-02-26 by xoseperez

rollercontainer, 2018-04-08:
Could this be used for a IR transmitter for smart meters?