xoseperez/espurna #532: Sonoff Bridge RF reboot every 70-80 min

albacete99, 2018-02-06:
I have Sonoff RF Bridge with 1.12.0 connected with Home Assistant vía mqtt.
He suspected that something was happening because sporadically the configured switches returned to their original state. I Configured a switch so that when I started it sent a mqqt code that was registered in HA. When reviewing the Log note that it restarts every 60-70 minutes.
Interestingly when accessing via Http also happened (this last not always)
I have uploaded to version 1.12.3 and I have also tried to change the power supply but it keeps restarting.
I do not know what else to do. Any suggestions?
Thanks in advance

xoseperez, 2018-02-07:
Do you have the chance to wire the device and open a console it (115200 bauds)? Leave it open and check when it's failing and the stack trace it outputs.

albacete99, 2018-02-07:
Is the same log/watchdog telnet and wire console?

xoseperez, 2018-02-08:
Yes, but if it's rebooting you will lose connection if using telnet.

albacete99, 2018-02-08:
Thanks @xoseperez .
Another question. Yesterday I was viewing with telnet (only few minutes) and I saw "Too many connections" 2-3 times. Is it normal? I only connected with telnet (no http).

albacete99, 2018-02-13:
Hi again.
I have connected during 4 hours by Telnet session....with no rebooting issue. But I disconnected and a few minnutes...rebooting.
Could it be a Deep Sleep problem?
Thanks

xoseperez, 2018-02-13:
ESPurna does not have deep sleep features built in. Try connecting again via telnet and type `info` to get the info about the device and last reboot reason and also `crash` to dump the last error stack trace. Copy them here, please.

albacete99, 2018-02-13:
I connected via Telnet but it didn't responsed when type info, crash,..
I have a Sonoff TH 10 and It works perfect with telnet+commands.
In https://github.com/xoseperez/espurna/wiki/Terminal you say "except for the Sonoff Dual and the Sonoff RF Bridge". Is this correct?
Thanks

xoseperez, 2018-02-13:
Ah yes. I forgot... shame. This is hard to debug. I'm adding code to dump info automatically on telnet connection when there is no terminal support. It's in the dev branch now.

albacete99, 2018-02-14:
Hi @xoseperez.
I tested dev branch and I update RF Bridge in Web (OTA). Then I connected with telnet and now start with this lines:
[352197] [TELNET] Client #0 connected
[352197]

[352197] [INIT] ESPURNA 1.12.4a
[352197] [INIT] xose.perez@gmail.com
[352198] [INIT] http://tinkerman.cat

[352198] [INIT] CPU chip ID: 0x9EF08D
[352198] [INIT] CPU frequency: 80 MHz
[352198] [INIT] SDK version: 2.1.0(deb1901)
[352198] [INIT] Core version: 2_4_0
[352198] [INIT] Core revision:
[352198]
[352199] [INIT] Flash chip ID: 0x144051
[352199] [INIT] Flash speed: 40000000 Hz
[352200] [INIT] Flash mode: DOUT
[352200]
[352200] [INIT] Flash sector size:     4096 bytes
[352200] [INIT] Flash size (CHIP):  1048576 bytes
[352201] [INIT] Flash size (SDK):   1048576 bytes /  256 sectors
[352201] [INIT] Firmware size:       501280 bytes /  123 sectors
[352202] [INIT] OTA size:            524288 bytes /  128 sectors
[352202] [INIT] SPIFFS size:              0 bytes /    0 sectors
[352202] [INIT] EEPROM size:           4096 bytes /    1 sectors
[352202] [INIT] Empty space:          16384 bytes /    4 sectors
[352202]
[352202] [INIT] BUILD_FLAGS:
[352203] [INIT] BOARD: ITEAD_SONOFF_RFBRIDGE
[352203] [INIT] SUP

I suppose this not a crash dump. Could a crash dump be launched just connect?
Thanks

xoseperez, 2018-02-14:
This dump is not complete. Some of the info is missing (after SUP...) , and there should be a crash dump also.

albacete99, 2018-02-14:
After SUP no more text appear un telnet.
I've tested 3-4 times but same result.
Thanks

closed 2018-07-14 by stale[bot]