xoseperez/espurna #533: Analog disappeared!

pcornill, 2018-02-07:
I am using Analog Pin in my project to read ADC value. In the last Espurna 1.12 release, the config disappeared. Is there something else instead?
Thanks for your great work!

xoseperez, 2018-02-07:
Now it's in the sensors.h file.

pcornill, 2018-02-07:
Yes, I noticed that but it seems the PIN configuration is gone in sensors.h and "hard coded" in sensors/AnalogSensor.h

Is there any particular reason for that?

       // Initialization method, must be idempotent
        void begin() {
            pinMode(0, INPUT);
        }

Thanks

xoseperez, 2018-02-08:
ESPurna only supports ESP8266 platform at the moment, and the ESP8266 only has 1 analog GPIO.

xoseperez, 2018-02-09:
Thanks to you 😄

closed 2018-02-09 by xoseperez

nqdanyb, 2018-06-28:
I have a heat resistor attached to the ADC, How to display the temperature with the unit
  Celsius on web interface and publish via mqtt to home-assistant? (sorry for my english) Thanks for all the help

xoseperez, 2018-06-28:
@nqdanyb if you enable ANALOG_SUPPORT in sensors.h you will get the raw ADC value. The temperature will depend on the thermistor you are using.