xoseperez/espurna #534: HC-SR04 in a sonoff SV

ervkane, 2018-02-07:
I have a HC-SR04, with a sonoff sv, both powered with 5%, but i don´t know if the HC-SR04 is supported in espurna. Here it´s run http://tinkerman.cat/sonoff-th10-th16-sensors-displays-actuators/

Thanks!

xoseperez, 2018-02-07:
Not at the moment. I'm tagging this as a possible enhancement. Thank you.

ruimarinho, 2018-03-19:
Or the RCW-0001 which works well with 3.3V :)

xoseperez, 2018-07-02:
Both HC-SR04 and RCW-0001 are supported since 1.12.6 setting HCSR04_SUPPORT=1. Please note that the HC-SR04 is a 5V devices whilst the RCW-0001 is a 3V3 (I recommend the later).

closed 2018-07-02 by xoseperez

ruimarinho, 2018-07-30:
@xoseperez would you be open if I replace this implementation with https://playground.arduino.cc/Code/NewPing which includes support for SR04, SRF05, SRF06, DYP-ME007, Parallax PING and RCW-0001 (3V3)?

It also has a built-in digital filter which I've found to be absolutely needed for error correction on these sensors.

Tofandel, 2019-10-05:
I don't see anything in the doc or the code related to HCSR04_SUPPORT, does it still works or was it changed?

How to set the pins?

Tofandel, 2019-10-05:
After a bit of research in the code

-DSONAR_SUPPORT=1 -DSONAR_TRIGGER={TRIGGER_PIN} -DSONAR_ECHO{ECHO_PIN}

I will update the doc