xoseperez/espurna #535: Sonoff POW reboots when switch is turned ON

pcornill, 2018-02-07:
Since I updated from 1.08 to 1.12 my Sonoff POW reboots few minutes after the switch is ON.
I suspect a correlation to wificonfig as I found several posts on the web related to this behavior.
I noticed that one of my Sonoff basic that was switched ON also rebooted when I tried to reboot my wifi router!

xoseperez, 2018-02-08:
What do you mean by "wificonfig"? I have two POWs here running without a problem.

pcornill, 2018-02-09:
I am referring to this thread:

https://github.com/arendst/Sonoff-Tasmota/issues/1034

I tested the last Espurna firmware (1.12) with all my Sonoff Basic, TH and POW and all of them reboot when I switch my wifi router off.

This is something new because I have a Sonoff with an older release that does not reboot.

So I guess something is wrong with my POW wifi and periodically it reboots.

xoseperez, 2018-02-09:
`wificonfig` is a setting in Tasmota, not ESPurna. Will try to reproduce this thou.

pcornill, 2018-03-27:
Hi, it seems to be gone with 1.12.4

xoseperez, 2018-04-04:
Closing then.. Feel free to reopen if needed.

closed 2018-04-04 by xoseperez